priv-kweihmann/oelint-adv #23: Add rule: sort alphabetically DEPENDS and RDEPENDS entries

jetm, 2019-10-14:
This rule is not based in OE style, but I have found useful to sort alphabetically `DEPENDS` and `RDEPENDS` to avoid duplicated entries in recipes with long `[R]DEPENDS` lists. Example:
```
DEPENDS_${PN} = "foo"
DEPENDS_${PN} += "bar"
```
It will become:
```
DEPENDS_${PN} = "bar"
DEPENDS_${PN} += "foo"
```

There is a case that would require more work to apply this rule. Bitbake allows put all the entries in one line, but with the time, that line will increase and will be harder to review it. That is why one recipe per line with `[R]DEPENDS` is better. Example:
```
DEPENDS = "foo bar"
```
Preferred style:
```
DEPENDS = "bar"
DEPENDS += "foo"
```
So, maybe this issue could become in two separated rules; one to sort the `[R]DEPENDS` entries and 2nd to convert on the one line `[R]DEPENDS` entries in separated lines. The latter could do both things, but found easier to do the former.

priv-kweihmann, 2019-10-15:
Hi @jetm 

thanks for your input. Actually I the following 2 rules in here

- sorting of [R]DEPENDS
- No duplicates in [R]DEPENDS

Regarding the sorting I personally like the following syntax more

```bitbake
DEPENDS += "\
            bar \
            foo \
           "
```

so you've got a the needed info in one block instead of spreading them across the recipe.
Is there any style reference for this at OE? Just grepping across the recipes I use, the "block" variant seems to be more common than the "multiline" variant - so what do you think? Would you go with that block-style as well?
I'll leave this issue here as it is and will create a new one for the "no duplicates"-case, as I see them separately.

jetm, 2019-10-15:
Hi @priv-kweihmann 

In OE is preferred the syntax you do like with `\`, that is why you can see it in most of all recipes. I understand its appeal: it's shorter and somehow cleaner. But I found in it some disadvantages that I explain below:

- It's forced to use `\` that is prone to error because normally people forget to put the `\` at the end, and that is not feel natural. 
- To get cleaner view it requires people indent it and in that change it gets easily mess up because sometimes people use tabs, instead of spaces or a mix. Sometimes use any number of spaces or tabs, and the uniformity gets easily broken.
- When the list goes longer, you will need to scroll up or down to know to which `[R]DEPENDS` that recipe belongs or if you're grepping you found harder to know where that recipe is at, because it will show only `recipe_foo \`.

priv-kweihmann, 2019-10-15:
I see your point. 
So I guess we need a configuration to address this kind of diverging requirements.

I'll see if I can find some time in the next days to code something useful...

commit 45d65d07 (2019-10-17): Rule: [R]DEPENDS should in alphabetical order

Closes #23