priv-kweihmann/oelint-adv #56: Output doesn't have color or a color option

FrazerClews, 2020-02-10:
I think this could be a really useful feature and something I am willing to work on in my own time if your want me too, the best way I've found is to use colorama for better portability if that's OK?

So red for errors, yellow for warnings, and green for info? Open to suggestions

priv-kweihmann, 2020-02-10:
Hi,

thanks for the input. Actually I'm not a big fan of coloring the console (as I mainly use terminals which doesn't support color codes properly), but if you like feel free to code such support.
I would insist that this feature is optional, which "no color" being the default.

A good starting point would clearly be https://github.com/priv-kweihmann/oelint-adv/blob/089b43492df0b2ca78e17df26c215e5e19ed90cc/oelint_adv/cls_rule.py#L19.

Come back to me if you help assistance or further details.

FrazerClews, 2020-02-10:
Yeah I agree with a --color option, personally i prefer color but I know plenty who don't. Thanks will keep that in mind while working on it

priv-kweihmann, 2020-02-11:
Merged with #63

closed 2020-02-11 by priv-kweihmann