priv-kweihmann/oelint-adv #112: Better handling of ${@} functions required

priv-kweihmann, 2020-03-09:
```
PACKAGES =+ "libpulsecore libpulsecommon libpulse libpulse-simple libpulse-mainloop-glib \
             pulseaudio-server pulseaudio-misc ${@bb.utils.contains('PACKAGECONFIG', 'dbus', 'pulseaudio-module-console-kit', '', d)}"
```
used with current code, leads to the effect, that `pulseaudio-server` isn't found

gstroz, 2020-03-09:
Idea: Would it make sense to use the upstream bitbake parser library and then have the linter work on the parsed data/AST? 

https://github.com/openembedded/bitbake/tree/master/lib/bb/parse

The upside would be getting the upstream parsing features for free. Downside would be reworking the linter and the licensing changes required.

priv-kweihmann, 2020-03-09:
@gstroz I agree that it would be easier. My idea of this linter is to provide a completely independent approach to bitbake, which includes an independent parser.
Otherwise there might be the change that something slips through. Also from what I see the parser itself changed a lot over the last years, so one would have to take care of versioning a.s.o.

Lately I had the idea to use a real lexer instead of regex+spaghetti code solution - but as always there are way more ideas than time available.

BTW if this project ever goes that way, wouldn't then make sense to include it directly into a bbclass to be used with bitbake - which would be a cool project - but for me the use case of this tool was always to lint a recipe, without the need to have any kind of bitbake/oe/YOCTO related tool installed.

Anyways thanks for the input - for this particular issue here, I just finished coding the patch, which works for me - so please do share your ideas as a separate issue, I'm really keen to know how your use cases look like.

commit a5d11940 (2020-03-09): Better handling of ${@} functions

Closes #112
closed 2020-03-09 by priv-kweihmann