priv-kweihmann/oelint-adv #118: Favor _append over +=

priv-kweihmann, 2020-03-14:
```
A ?= "1"
B ?= "1"
A += "2"
B_append = " 2"
```
turns into
```
A = "2"
B = "1 2"
```
which is a confusing behavior of bitbake - therefore `_append` should be the preferred method

closed 2020-03-15 by priv-kweihmann

rossburton, 2020-03-24:
Globally preferring _append over += is bad form though, so I'd say this is an over-zealous warning.

priv-kweihmann, 2020-03-24:
You're maybe right - it is a bit overly picky. Guess I will limit it to known variables with weak assignments.
Or do you have different suggestions?

Just created the rule after I stumbled upon some thingy described in the ticket

rossburton, 2020-03-24:
Not sure you can generalise it to a universal rule.  += on a variable that was ?= isn't ideal if you can manage to parse that out.

priv-kweihmann, 2020-03-24:
Well I'm aware of that - outcome of the tool shall be interpreted as advice not as hard facts.
Guess I'll go that way then to apply this rule only to variables that have obvious weak assignments in the known scope.

Would you do me a favor and create an issue and briefly describe your concerns - Thanks

commit f29f30d8 (2020-03-15): Favor _append over +=

Closes #118