priv-kweihmann/oelint-adv #165: PACKAGES are not fully expanded

priv-kweihmann, 2020-06-02:
```
TZ_PACKAGES = " \
    tzdata-core tzdata-misc tzdata-posix tzdata-right tzdata-africa \
    tzdata-americas tzdata-antarctica tzdata-arctic tzdata-asia \
    tzdata-atlantic tzdata-australia tzdata-europe tzdata-pacific"
PACKAGES = "${TZ_PACKAGES} ${PN}"
```
PACKAGES do not contain any of the mentioned packages from above while checking

closed 2020-06-02 by priv-kweihmann

sidsharma, 2020-08-24:
Hello, PACKAGE_BEFORE_PN does not seem to be considered in calculation of the PACKAGES yet. I checked 1.23.1.

priv-kweihmann, 2020-08-24:
Yes, full support was added with 1.24.0

sidsharma, 2020-08-25:
I just checked on 1.24.0, PACKAGE_BEFORE_PN is still not respected it seems.

```
recipe.bb:68:error:oelint.vars.specific:'FILES' is set specific to ['extra-pkg'], but isn't known from PACKAGES, MACHINE or resources
recipe.bb:68:error:oelint.vars.pkgspecific.FILES:Variable FILES is package-specific and therefore it should be FILES_${PN}
```

I have also seen issues with other variables such as INITSCRIPT_NAME when using overrides based on packages added via PACKAGE_BEFORE_PN.

priv-kweihmann, 2020-08-26:
Okay I'll check that

commit ae79233e (2020-06-02): PACKAGES are not fully expanded

Closes #165