priv-kweihmann/oelint-adv #183: Provide the recipe parser as its own library

pombredanne, 2020-10-07:
As a follow up to this discussion https://github.com/nexB/scancode-toolkit/issues/1243 it would be great to be able to use the parser standalone possibly in it own library.

Why? The bitbake parser cannot work in practice for files analysis as it needs a whole context of correct files to be happy, as shown here: https://github.com/nexB/scancode-toolkit/issues/1243#issuecomment-704869940

In contrast the parser here can work nicely with a single file and has been designed for use in analysis tools.

Here is a snippet on how to use it alone today:
```Python
from oelint_adv.cls_stash import Stash
from oelint_adv.parser import get_items

class Args(object):
    def __init__(self):
        self.quiet = True

s = Stash(args=Args())
f = 'dropbear_2.1.bb'
items = get_items(s, f)
for item in items:
  for key, value in item.GetAttributes().items():
    print(key, ':', value)

Line : 1
Raw : SUMMARY = "A lightweight SSH and SCP implementation"

Links : []
Origin : dropbear_2.1.bb
InFileLine : 1
IncludedFrom : []
VarName : SUMMARY
SubItem : 
...
```

pombredanne, 2020-10-08:
@priv-kweihmann I see you started https://github.com/priv-kweihmann/oelint-parser ... that's great! 
Is there something I can do to help?

priv-kweihmann, 2020-10-08:
Not really - will push a WIP soon (maybe Saturday or early next week) so you can test it, that would be helpful and highly appreciated

priv-kweihmann, 2020-10-08:
@pombredanne alright, earlier than expected, but you can find the initial implementation at https://github.com/priv-kweihmann/oelint-parser/tree/WIP_initial.
Please give it a try and report any bugs or improvements back to the issue board of that repo.

pombredanne, 2020-10-09:
And I entered https://github.com/priv-kweihmann/oelint-parser/issues/1 and we can track these there now.
I am closing this ticket as this is solved there.
Again thank you ++ :bow:

closed 2020-10-09 by pombredanne

priv-kweihmann, 2020-10-16:
@pombredanne @chinyeungli Initial release is out