priv-kweihmann/oelint-adv #190: Linting is unusably slow

skycaptain, 2020-11-12:
First of all, thank you for this tool. We would like to use it to lint a rather large codebase (1000+ Bitbake files). However, we find it unreasonably slow, especially when being used as a pre-commit hook tool. In average linting a single files takes around 1-2 seconds on a recent and decent equipped Macbook Pro running Python 3.9.0. To understand this yourself, you could try to run linting on the complete meta-openembedded.

I profiled this codebase and found `oelint_parser.cls_item.Item.GetAttributes`, where `inspect` is called inside a loop to be function which takes up the most time during execution. Is there any specific reason to use inspect? E.g. you could use something like, which is much faster:

```python
    def GetAttributes(self):
        """Get all public attributes of this class

        Returns:
            dict -- all public attributes and their values
        """
        
        res = {}
        for _ in dir(self):
            try:
                attr = getattr(type(self), _)
            except AttributeError:
                continue  # Skip dunder attributes
            else:
                if isinstance(attr, property):
                    res[_] = getattr(self, _)

        return res
```

priv-kweihmann, 2020-11-12:
Yeah you're right - this is so far not optimized for performance and yeah the mentioned code path is pretty slow. Could you craft a PR for that, as I would be really interested in speeding it up again?

skycaptain, 2020-11-12:
Of course, will do so. Have you ever considered using [attrs](https://www.attrs.org/en/stable/) for class abstraction?

priv-kweihmann, 2020-11-12:
Cool. 
And I actually thought about it, but I want to keep the parser as dependency free as possible, so I decided against using it for now.

priv-kweihmann, 2020-11-13:
oelint-parser 1.0.6 released - closing this one here - again thanks a lot

closed 2020-11-13 by priv-kweihmann