priv-kweihmann/oelint-adv #202: file globbing

HerrMuellerluedenscheid, 2021-03-09:
Hey, 
Thanks for an awesome tool! Does `oelint-adv` allow globbing files or walking through a file tree of a meta layer?

priv-kweihmann, 2021-03-09:
Not yet - for now every bb/bbappend file has to be passed via CLI explicitly.
I usually do `find <dir> -name "*.bb" -o -name "*.bbappend" | parallel oelint-adv --quiet {}` when I want to run it on a complete layer.

Most of the time I personally use the tool on PR/MR so I already got a change set at hand which should be scanned - so no need to traverse a tree.
Another usage is with meta-sca, where every recipe that is actually build is scanned - there all the files used are provided by bitbake itself.

Can you somehow elaborate what you would like to do and what should be the expected outcome - maybe one could implement something that suits all the mentioned use cases

priv-kweihmann, 2021-06-02:
@HerrMuellerluedenscheid any comment on that? ^^^