priv-kweihmann/oelint-adv #203: False positive with ordering

rossburton, 2021-03-09:
If I oelint optee-test:

http://git.yoctoproject.org/cgit/cgit.cgi/meta-arm/tree/meta-arm/recipes-security/optee/optee-test_3.11.0.bb

```
meta-arm/recipes-security/optee/optee-test.inc:33:warning:oelint.task.order.do_compile:'do_compile' should be placed before 'do_install'
```

But do_compile is before do_install.

priv-kweihmann, 2021-03-09:
could you give it a try with the latest version 2.2.2 - I didn't encountered it anymore

rossburton, 2021-03-10:
Yes, 2.2.2 fixes.

closed 2021-03-10 by rossburton

robertsemaskevic, 2021-03-11:
It seems that the issue is still present with multiple bbappends from different layers but for the same package.
Minimum steps to reproduce:
```
├── meta-foobar
│   ├── conf
│   │   └── layer.conf
│   └── recipes-foobar
│       └── foobar.bbappend
└── meta-foobar2
    ├── conf
    │   └── layer.conf
    └── recipes-foobar
        └── foobar.bbappend
```
```
[21:45:33] $ cat meta-foobar/recipes-foobar/foobar.bbappend
do_install_prepend() {
    echo "foobar install"
}
[21:45:37] $ cat meta-foobar2/recipes-foobar/foobar.bbappend
do_compile_prepend() {
    echo "foobar compile"
}
```
Also, the linter output is random on every execution:
```
[21:45:53] $ oelint-adv meta-foobar/recipes-foobar/foobar.bbappend meta-foobar2/recipes-foobar/foobar.bbappend --quiet --color --suppress="oelint.var.bbclassextend"
.../meta-foobar2/recipes-foobar/foobar.bbappend:1:warning:oelint.task.order.do_compile:'do_compile' should be placed before 'do_install'
[21:47:07] $ oelint-adv meta-foobar/recipes-foobar/foobar.bbappend meta-foobar2/recipes-foobar/foobar.bbappend --quiet --color --suppress="oelint.var.bbclassextend"

[21:47:08] $ oelint-adv meta-foobar/recipes-foobar/foobar.bbappend meta-foobar2/recipes-foobar/foobar.bbappend --quiet --color --suppress="oelint.var.bbclassextend"

[21:47:10] $ oelint-adv meta-foobar/recipes-foobar/foobar.bbappend meta-foobar2/recipes-foobar/foobar.bbappend --quiet --color --suppress="oelint.var.bbclassextend"

[21:47:11] $ oelint-adv meta-foobar/recipes-foobar/foobar.bbappend meta-foobar2/recipes-foobar/foobar.bbappend --quiet --color --suppress="oelint.var.bbclassextend"
.../meta-foobar2/recipes-foobar/foobar.bbappend:1:warning:oelint.task.order.do_compile:'do_compile' should be placed before 'do_install'
```

The above valid for oelint-parser 1.0.10, oelint-adv 2.2.2, python 3.6|3.9.

priv-kweihmann, 2021-03-11:
@robertsemaskevic could you please open a new bug for the non deterministic behavior - thx