priv-kweihmann/oelint-adv #204: Confused over variable overriding

rossburton, 2021-03-09:
If I oelint the optee recipes:

http://git.yoctoproject.org/cgit/cgit.cgi/meta-arm/tree/meta-arm/recipes-security/optee

```
optee-examples.inc:42:error:oelint.var.override:Variable 'PACKAGE_ARCH' is set by optee-test.inc,optee-os.inc,optee-examples.inc
```

Yes.  Why is it telling me this? They are independent recipes.

priv-kweihmann, 2021-03-09:
Basically it says that `PACKAGE_ARCH = "${MACHINE_ARCH}"` is set by all of the 3 mentioned file, which are all included in the same recipe - for me `PACKAGE_ARCH` might be uncritical, but this is a more general rule, which should catch those nifty overrides from inc-files.
But yeah I agree wording could be improved - any ideas?

priv-kweihmann, 2021-03-09:
but let me check that in detail...

priv-kweihmann, 2021-03-09:
Deng, I see and let me guess you did run the tool like `oelint-adv <options> *.bb`, right?
In that case yes, this is a false positive

rossburton, 2021-03-09:
Yes, I did *.bb.

commit a8bd9ffc (2021-03-09): main: group files

in case multiple unrelated bb files are passed via CLI, group them into
linked groups (bb + matching bbapends) and pass them separately to the
paser. Lone bbappends are handled as groups of their own.
This avoid having settings from shared inc-files spread in the stash
and leading to false postive warnings.

Closes #204
closed 2021-03-09 by priv-kweihmann