priv-kweihmann/oelint-adv #217: DEFAULT_PREFERENCE shouldn't be set in recipe?

skycaptain, 2021-05-27:
Running latest `oelint-adv` over my metalayer gives me:

```
[...]warning:oelint.vars.bbvars.DEFAULT_PREFERENCE:Variable 'DEFAULT_PREFERENCE' should be set on a disto/layer or local.conf level, not in a recipe
```

However, the current [docs](https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-DEFAULT_PREFERENCE) state:

> The most common usage of this is variable is to set it to "-1" *within a recipe* for a development version of a piece of software.

Also, multiple openembedded recipes are using `DEFAULT_PREFERENCE` in recipe files, but I havn't seen any example of usage in distro or layer conf. How would I set `DEFAULT_PREFERENCE` here for a specific version only? Could you please explain pros and cons of setting `DEFAULT_PREFERENCE` in distro or layer conf and maybe provide example code?

priv-kweihmann, 2021-05-27:
Okay, that sounds like an easy fix https://github.com/priv-kweihmann/oelint-parser/blob/main/oelint_parser/const_vars.py needs to be removed and the tests here adjusted - will spin up the necessary changes over the weekend.
Thx for reporting (and reading the manual :-) )

skycaptain, 2021-05-27:
Thanks!

> [...] and reading the manual :-) )

They put much effort into their docs and they are a very good read. I still miss more examples and best-practices on where to put things, but `DEFAULT_PREFERENCE` was clear.

priv-kweihmann, 2021-05-28:
After giving it some thought, I will rewrite the whole configuration of know constants (variables, functions, etc.) so this will take longer than initially expected