priv-kweihmann/oelint-adv #218: Sorting confused with includes

skycaptain, 2021-05-27:
One very common use-case is to put  the essence of a recipe in a include (`.inc`) file and put only `PV`, `SRCREV` and `SRC_URI` in the recipe [1]:

my-library.inc:

```bitbake
SUMMARY = "..."
DESCRIPTION = "...
AUTHOR = "..."
HOMEPAGE = "..."
BUGTRACKER = "..."
SECTION = "..."
LICENSE = "..."
LIC_FILES_CHKSUM = "..."

DEPENDS += "..."

SRC_URI = "..."

inherit cmake

EXTRA_OECMAKE += "..."

BBCLASSEXTEND += "nativesdk"
```

my-library_1.2.3.bb:

```bitbake
require my-library.inc

SRCREV = "..."
```

However, latest oelint-adv complains about the sorting of some variables:

```plain
[...]:warning:oelint.var.order.SRCREV:'SRCREV' should be placed before 'BBCLASSEXTEND'
```

As I still want to use the sorting check for single-file recipes, I would not want to suppress the check.

---
[1]: The reason for this is that Yocto uses versions in recipe filenames and upgrading recipes with also large changes in the recipe file itself often breaks Git history. Out-sourcing the essence of the recipe to a include without the version in its filename is one way to circumvent this. This method is also widely used by Yocto's own recipes.

priv-kweihmann, 2021-05-27:
Hmmm, I get your point - will check how to fix that properly, but it might need some adjustments in your recipe (will let you know once I found a good way)

priv-kweihmann, 2021-05-28:
I guess the best way will be to limit the ordering rules (variables and function) to the scope of each individual file - added that to my agenda

closed 2021-05-30 by priv-kweihmann

skycaptain, 2021-05-31:
Awesome. Works like a charm! Thanks.

commit 01d543b5 (2021-05-30): Limit *.order rules to individual files

when checking *.order rules (task and variables), only check the
order in the scope of the individual file.

This allows the use of include files before or after the
task/variable assignment

Closes #218