priv-kweihmann/oelint-adv #220: How to mute check for single suggested variable?

skycaptain, 2021-05-31:
I find the `oelint.var.suggestedvar` really helpful, as it keeps code quality high and "forces" developers to add proper metadata. While IMHO all suggested variables make sense, I would argue that `CVE_PRODUCT` is often not applicable to recipes of non-public software, i.e. enterprise embedded software. While I would still use the check for all other variables it would be great to be able to mute the check for a single variable. I tried to use `--suppress oelint.var.suggestedvar.CVE_PRODUCT`, however this muted all suggestions, not only `CVE_PRODUCT`.

skycaptain, 2021-05-31:
I guess this could be done by creating a rule class for each suggested var, possibly also dynamically. Shall I open a PR?

priv-kweihmann, 2021-06-01:
yes, that could be done, but this here sounds more like a bug to me - I will check on that later

priv-kweihmann, 2021-06-01:
Yes it's a bug :-(

closed 2021-06-01 by priv-kweihmann

skycaptain, 2021-06-02:
Once again thank you @priv-kweihmann, I really appreciate your work and engagement. I've tested 2.5.1; works like a charm!

skycaptain, 2021-06-02:
I'm now also able to disable certain other "subrules" and streamline everything.

Unfortunately, your recent change only seems to work with `--suppress` from the CLI but not with rulefiles. I'm also not able to just set `"oelint.var.suggestedvar.CVE_PRODUCT": "info"` inside the rulefile and use `--noinfo` with the CLI; oelint.var.suggestedvar.CVE_PRODUCT infos are still reported.

priv-kweihmann, 2021-06-02:
@skycaptain yep, haven't looked at that, but it's likely to be like you reported. Could you please open another ticket and I will address that in a Bugfix release - thx

commit bf0ce1ce (2021-06-01): suppress: fix suppression of sub-IDs

instead of suppressing findings by not loading the rules,
suppress the findings at creation of the instance, as we otherwise
can't distinguish between one suppressed appendix id and
another unsuppressed.

Closes #220