priv-kweihmann/oelint-adv #238: oelint.vars.listappend message is switched up?

Rahix, 2021-09-15:
I think the message for `oelint.vars.listappend` is the wrong way around... The lint checks for

```python
if not i.VarValue.startswith("\" ")
```

and then instructs me that

> Append to list should **end** with a blank

The same but reversed is true for prepending...  Am I missing something or should this message be updated?

Additionally, I think the lint should not trigger for `FILESEXTRAPATHS:prepend`?

priv-kweihmann, 2021-09-15:
Yes, please do so - and I kind of suspected that this is somehow mixed up and was lacking the time to have a deeper look lately

Rahix, 2021-09-15:
Okay, I think the `FILESEXTRAPATH` is a bigger thing: There are quite a few variables which are colon separated and I think it makes sense to make the lint also check all those.  Now I am not quite sure how this works, I see there are constants defined in `oelint_parser` for e.g. mandatory variables.  Should I be doing the same here?  I.e. add a new list for all the colon separated and space separated lists?

Rahix, 2021-09-15:
(in any case, I'll just make it silent for `FILESEXTRAPATHS` now to get this fix in as quickly as possible and them come back later to implement the full linting of it)

PR #240 (2021-09-15): listappend: Actually fix the mixup in the lint message
For appending, you need to start with a blank, for prepending you need to end with a blank.

Followup fix for 52e7ddbc9737 ("Fix mixup in error message").

Additionally, fix this lint triggering false-positive for `FILESEXTRAPATHS`.

Fixes #238.
closed 2021-09-15 by priv-kweihmann