priv-kweihmann/oelint-adv #244: Error message for wrong `Inappropriate` upstream status is confusing

Rahix, 2021-09-17:
Suppose a patch with a line like this:

```text
Upstream-Status: Inappropriate [specific to our board]
```

the linter will generate the following error message:

```text
foo.bb:13:warning:oelint.file.upstreamstatus:Upstream-Status in 'foo.patch' doesn't pick from valid classifiers Accepted,Backport,Denied,Inappropriate,Pending,Submitted
```

This is confusing because it certainly does use a valid classifier.  Only when looking at the lint implementation does one see that the problem is with the text in `[]`.  I think the `[]` text should be checked by a separate lint to a) provide a more clear error message and b) allow disabling this sub-check in case projects adopt different `Inappropriate` reasons...

Alternatively, I think it would also be okay to not check the text inside `[]` at all as the chance of someone adding the `Upstream-Status: Inappropriate` tag without writing a sensible reason is slim.

priv-kweihmann, 2021-09-17:
At this point I'd like just point at the commit message guide of open-embedded :) - because that one is overly picky about all kinds of stuff - the rule is just reflecting on that. I agree that the error message could point to the correct violation.
Fun fact even in oe-core more than one patch is not sticking to the guideline in total.

I would be okay in creating a constant set of what is acceptable for the [] part - so there's a standard set (the "pure" commit guideline) but it can be extended to what is acceptable in your project.

If that's okay for you I can work on that in the next few days

priv-kweihmann, 2021-09-17:
What I wouldn't allow is a completely empty [] (or not existing) - I've seen way too many patches that gave no hint why it should be inappropriate, making it difficult to get rid of them, piling up technical debt in a code base. From my perspective in case one is unsure just pick Pending state.

And judging from the chats I had with the core oe team, there is rather the tendency to make this more strict than allowing some sort of freeform declaration

Rahix, 2021-09-20:
Alright, makes sense.  I think the most important part for me would be having a clear error message.  Otherwise people don't know what they need to change.  Having the list of what's acceptable might be nice, but to be honest, if upstream OE doesn't want to allow it, I don't see much benefit from having it here...

priv-kweihmann, 2021-09-20:
Okay, I will split the message part off into separate rules - and maybe, maybe allow some customization to them (depending on how complicated that change will get)

commit ef7f06dc (2021-09-22): rules: split Inappropriate msg rule into new rule

and provide a better error message,
so people have a clou where to look at

Closes #244
closed 2021-09-22 by priv-kweihmann