priv-kweihmann/oelint-adv #245: `type` fetcher option should be allowed in linux-yocto recipes

Rahix, 2021-09-24:
Hi,

I'm getting a false positive about `SRC_URI` entries in recipes derived from `linux-yocto`:

```text
linux-[...].bbappend:9:warning:oelint.vars.srcurioptions:Option 'type' is not known with this fetcher type
```

This should be allowed as it is used for kernel metadata as detailed in ["Metadata Outside the Recipe-Space"](https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#metadata-outside-the-recipe-space).

I am not sure how to handle this.  AFAIK, this is only valid in `linux-yocto` based recipes, so the lint should still be active everywhere else?  How would I check for this in a lint implementation?

priv-kweihmann, 2021-09-24:
Thanks for another finding.
Regarding a possible fix, I'd say somewhere here https://github.com/priv-kweihmann/oelint-adv/blob/586c2412914f9637e245d1b5268a830f300679b2/oelint_adv/rule_base/rule_var_src_uri.py#L141 a(n exception would make sense.

in pseudo code smt like 
```
if (inherits linux-yocto) && (_url["scheme"] == 'git') && (k == 'type'):
   continue
```

can you code such a fix? or should I take it?

Rahix, 2021-09-27:
> can you code such a fix? or should I take it?

I'd give it a shot :)

> > How would I check for this in a lint implementation?

Sorry, I should have phrased this more clearly... I was looking for help with how I can do the `if (inherits linux-yocto)` check robustly.  Think, for example, that the `inherit linux-yocto` statement is often in an `.inc` file which is then `require`d in the actual recipe.  Is it even possible to check for this without huge effort?

priv-kweihmann, 2021-09-27:
> Sorry, I should have phrased this more clearly... I was looking for help with how I can do the `if (inherits linux-yocto)` check robustly. Think, for example, that the `inherit linux-yocto` [...]

Sure `include`/`require` statements are automatically expanded (but the include has to be in the same layer, that's the only limitation), so just doing 

```
        items = stash.GetItemsFor(filename=_file, classifier=Variable.CLASSIFIER,
                                  attribute=Variable.ATTR_VAR, attributeValue="inherit")
        for i in items:
            for subi in [expand_term(stash, _file, x) for x in i.get_items() if x and x != INLINE_BLOCK]:
                if subi == "linux-yocto":
                    <do something>
 ```
 should do it - you can look at https://github.com/priv-kweihmann/oelint-adv/blob/master/oelint_adv/rule_base/rule_vars_improperinherit.py for more inspiration - happy coding ;-)

Rahix, 2021-10-14:
I agree, it feels like an ugly hack to me just as well...

The project which I am targeting with this is unfortunately not (yet!) public, but it isn't actually too difficult to show:  Essentially, we have a [`meta-mainline-common`](https://source.denx.de/denx/meta-mainline-common) layer with a [`linux-stable_5.10.bb`](https://source.denx.de/denx/meta-mainline-common/-/blob/4b12a4095dfe80900cd1578d4885b06d83223496/recipes-kernel/linux/linux-stable_5.10.bb) recipe.  Ontop of this, a device specific bsp-layer adds kmeta for a certain hardware in a bbappend.  As an example, it might look like this:

```sh
# linux-stable_5.10.bbappend
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

BPV := "${@'.'.join(d.getVar('PV').split('.')[0:2])}"
KBRANCH:some-custom-board ?= "linux-${BPV}.y"
KMACHINE:some-custom-board ?= "some-custom-board"
COMPATIBLE_MACHINE:some-custom-board = "(some-custom-board)"

SRC_URI:append:some-custom-board = " \
	file://${BPV}/some-custom-board;type=kmeta;destsuffix=${BPV}/some-custom-board \
	"
KERNEL_FEATURES:some-custom-board = " some-custom-board.scc "
```

As this bbappend is in a completely different layer, there is not way to trace back to an include or inherit statement, as far as I can tell...

closed 2021-10-14 by priv-kweihmann