priv-kweihmann/oelint-adv #283: Some .oelint.cfg options ignored

offa, 2021-11-16:
I have added a `.oelint.cfg` file, but some of it's options are ignored. Even though it's in the project root (from where oelint is run) or `OELINT_CONFIG` is set to the file path.

### Example config

```ini
[oelint]
supress = oelint.vars.homepageping
exit-zero = True
# quiet = True # The quiet option works
```

### Reproduction

1. Install *oeline-adv* from Git (I'm using master at a2dbed2)
2. Add `.oelint.cfg` file as above
3. Run *oelint-adv* on a file reporting an issue
4. Options are ignored: non-zero is returned, suppressed rule is listed

priv-kweihmann, 2021-11-16:
Will have a look. Are the stated options the only ones that don't work?

offa, 2021-11-16:
Those are the ones I've used so for. I did some quick testing: `color`, `relpaths `and `noid `are working too.

Additionally, it seems options used in the config must not use `-` but `_`. Using `exit-zero` is ignored, but `exit_zero` works.

offa, 2021-11-16:
Regarding suppressions: Should a suppressed rule appear in the loaded rules list? Respectively, does a loaded rule imply it's execution?

offa, 2021-11-16:
@priv-kweihmann probably this isn't a bug. It seems suppressed rules appear as loaded, but aren't actually executed and options used in the config file must use `_` instead of `-`.

priv-kweihmann, 2021-11-16:
That's most likely - will work on a fix soon

priv-kweihmann, 2021-11-18:
> Regarding suppressions: Should a suppressed rule appear in the loaded rules list? Respectively, does a loaded rule imply it's execution?

Well the suppression aims at the findings, thus all loaded rules are actually run, but findings aren't presented

offa, 2021-11-18:
I see, so I misinterpret the output and it's working correctly. 

The only issue left are the `-` vs. `_`, but adding a note to docs should solve this.

closed 2021-11-18 by priv-kweihmann

offa, 2021-11-18:
@priv-kweihmann Looks good, options like `exit-zero`. Thank you! 😄

commit 36ae0970 (2021-11-18): cli: escape options with dashes from cfgfile

as there is a hidden translation from a-b to a_b within argparse,
so we have to the same for the options that we load
from the configuration file.
Add tests to avoid future regressions

Closes #283