priv-kweihmann/oelint-adv #285: False positive from oelint.var.multiinclude

Rahix, 2021-12-02:
Hi!

I found another oddity where I am unsure what is going on...  I have a custom U-Boot recipe, which is built up similar to the upstream one.  With this, I am getting very strange linter errors.  It can be reproduced by just 3 files like this:

```
───────┬───────────────────────────────────────────────────────────
       │ File: u-boot-rahix.bb
───────┼───────────────────────────────────────────────────────────
   1   │ require u-boot-rahix-common.inc
   2   │ require recipes-bsp/u-boot/u-boot.inc
───────┴───────────────────────────────────────────────────────────

───────┬───────────────────────────────────────────────────────────
       │ File: u-boot-rahix-common.inc
───────┼───────────────────────────────────────────────────────────
   1   │ require recipes-bsp/u-boot/u-boot-common.inc
───────┴───────────────────────────────────────────────────────────

───────┬───────────────────────────────────────────────────────────
       │ File: u-boot-rahix-tools.bb
───────┼───────────────────────────────────────────────────────────
   1   │ require u-boot-rahix-common.inc
   2   │ require recipes-bsp/u-boot/u-boot-tools.inc
───────┴───────────────────────────────────────────────────────────
```

When linting just one of the files, all looks as expected.  When linting all at the same time however, for example with:

```bash
oelint-adv --quiet --color --relpath u-boot-rahix.bb u-boot-rahix-tools.bb
```

I get some very odd errors (only relevant ones are shown):

```text
u-boot-rahix.bb:1:warning:oelint.var.multiinclude:'u-boot-rahix-common.inc' is included multiple times
u-boot-rahix-common.inc:1:warning:oelint.var.multiinclude:'recipes-bsp/u-boot/u-boot-common.inc' is included multiple times
```

Even more, when the recipes contain non-trivial content, I additionally get errors like this one:

```text
u-boot-rahix-common.inc:2:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
```

while the file certainly did not contain multiple empty lines...

Any idea what is going on here?

priv-kweihmann, 2021-12-02:
hmmmm, that looks like something I already fixed a while ago - I'll look into that in the next few days.
Likely the grouping of file goes wrong...

What you could try is to put a version into the filenames of the bb-files.

u-boot-rahix.bb -> u-boot-rahix_git.bb
u-boot-rahix-tools.bb -> u-boot-rahix-tools_git.bb

and rerun it with the same settings... I wouldn't be surprised if that actually fixes the false positives (and gives me a hint where to look for the issue)

priv-kweihmann, 2021-12-02:
And I had the right hunch, fix incoming...

closed 2021-12-02 by priv-kweihmann

Rahix, 2021-12-02:
Thanks for the prompt fix!

commit 5f8ea5e2 (2021-12-02): main: fix grouping for non versioned recipes

if a recipe doesn't contain a version info (_<version>) then
in before the grouping didn't work.
Fix that by explicitly checking a potentially non exisiting version

Closes #285