priv-kweihmann/oelint-adv #293: Rule to check correct SRC_URI

HerrMuellerluedenscheid, 2022-02-02:
I was using [meta-raspberrypi](https://github.com/agherzan/meta-raspberrypi) at `77190af02d48adc2b28216775e6318e9eeda571c` which has the following `SRC_URI` in `recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb`:

    `SRC_URI = "git://github.com/RPi-Distro/firmware-nonfree"`

Issue: `protocol=https` was missing and was fixed in a subsequent commit. Apparently, the wrong URL is correctly translated (probably on the destination host) to https://github.com/RPi-Distro/firmware-nonfree. In more secured networks (such as the one in our institute) however, the bitbake build fails at this recipe because the request times out. I assume, there is an iptables rule which blocks the malformatted URL.

Solution would be a linting rule the uses regex to identify if the URL either starts with `http[s]://` or `git@` or **`git://`+ has `protocol=http[s]`** defined.

priv-kweihmann, 2022-02-02:
Sounds like a good idea - can you provide a few bad test cases data samples? then I'll take it from there

HerrMuellerluedenscheid, 2022-02-02:
> Sounds like a good idea - can you provide a few bad test cases data samples? then I'll take it from there

Yes. This hit hard on me a couple of times also from peer-reviewed recipes ;) I'll have a look into our project history to see if I can find the previous cases.

priv-kweihmann, 2022-02-16:
@HerrMuellerluedenscheid please have a look if that solves your issues.
I checked it against poky master and meta-openembedded master with good results

commit 40afc46f (2022-02-16): Rule to check correct SRC_URI

check if git comes with some needed and some highly recommended
parameters

Closes #293
closed 2022-02-16 by priv-kweihmann