priv-kweihmann/oelint-adv #297: Includes are included multiple times with 3.7.2?

skycaptain, 2022-03-20:
The changes introduced with 3.7.2 are giving strange errors when running against recipes with includes:

my-library.inc:

```bitbake
SUMMARY = "..."
DESCRIPTION = "...
AUTHOR = "..."
HOMEPAGE = "..."
BUGTRACKER = "..."
SECTION = "..."
LICENSE = "..."
LIC_FILES_CHKSUM = "..."

DEPENDS += "..."

SRC_URI = "..."

inherit cmake

EXTRA_OECMAKE += "..."

BBCLASSEXTEND += "nativesdk"
```

my-library_git.bb:

```bitbake
require my-library.inc

SRCREV = "..."
```

results in:

```plain
my-library_git.bb:14:warning:oelint.var.multiinclude:'hello-library.inc' is included multiple times
my-library.inc:10:warning:oelint.var.order.HOMEPAGE:'HOMEPAGE' should be placed before 'BUGTRACKER'
my-library.inc:12:warning:oelint.var.order.SECTION:'SECTION' should be placed before 'LICENSE'
my-library.inc:13:warning:oelint.var.order.LICENSE:'LICENSE' should be placed before 'LIC_FILES_CHKSUM'
my-library.inc:15:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:16:warning:oelint.vars.duplicate:Item 'fmt' was added multiple times to DEPENDS
my-library.inc:20:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:23:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:25:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:26:warning:oelint.var.multiinherit:'cmake' is included multiple times
my-library.inc:27:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:34:warning:oelint.newline.consecutive:Consecutive blank lines should be avoided
my-library.inc:7:warning:oelint.var.order.SUMMARY:'SUMMARY' should be placed before 'DESCRIPTION'
my-library.inc:8:warning:oelint.var.order.DESCRIPTION:'DESCRIPTION' should be placed before 'AUTHOR'
my-library.inc:9:warning:oelint.var.order.AUTHOR:'AUTHOR' should be placed before 'HOMEPAGE'
```

It seems like the include is included multiple times. 3.7.1 is fine and does not report this error.

priv-kweihmann, 2022-03-21:
Fixed it in a new release - by the way 3.7.x is a little outdated - current release 3.11.x, so you might want to consider an update

closed 2022-03-21 by priv-kweihmann

skycaptain, 2022-03-21:
Thank you very much for the quick response. Will try it asap. Yes, 3.7.x is outdated, however the latest version also showed this issue. It was just to mention the version when the issue appeared first.

skycaptain, 2022-03-24:
The solution gives me `ValueError: '/foo/bar/baz/my-recipe.bb' is not in list` when sorting `res` at the end of the `group_files` function as commented here: https://github.com/priv-kweihmann/oelint-adv/commit/ee585dca16ae07fe066c8c1154ea733e18e59a12#r69469006

commit ee585dca (2022-03-21): main: fix grouping of recipes

now different versions of the same recipe are treated
as individual groups.
Addtionally apply optional bbappends correctly to those
groups and not quit on the first best match.

Example input

test_1.2.3.bb
test_git.bb
test_1.%.bbappend
test_%.bbappend

will end up in the following groups

test_git.bb, test_%.bbappend
test_1.2.3.bb, test_%.bbappend, test_1.%.bbappend

which should rule out a lot of false positives

Closes #297