priv-kweihmann/oelint-adv #305: Upstream Status false positive

offa, 2022-07-06:
_"Patch '[…]' should contain an Upstream-Status entry"_ is reported on patches with a valid entry, eg:

```
Upstream-Status: Inappropriate [other]
```

There's seems no difference between reasons.

On oelint-adv 3.13.0 it works correctly and reports only those files with lack such entries.

A quick bisect showed 63ba6d667bedaa7a7285432f9a18454d13b6ed1f (first commit after 3.13 release).

##### Usage:

```
oelint-adv --quiet ./<Path to .bb file>
```

##### System:

- oelint-adv 3.15.1
- oelint-parser 2.5.0
- Python 3.10.5

priv-kweihmann, 2022-07-06:
Mhmm I tried to reproduce it without success - I don't get a false positive here.
My guesses would be encoding of the patch file, or tabs instead of spaces - could you please attach a trimmed down patch file that is reporting this error

offa, 2022-07-06:
[Here's an example](https://github.com/priv-kweihmann/oelint-adv/files/9052584/example.tar.gz) recipe with patch which reports:

```
oelint-adv --quiet audit/audit_3.0.8.bb 
[…]/audit/audit_3.0.8.bb:5:info:oelint.file.upstreamstatus:Patch 'Fixed-swig-host-contamination-issue.patch' should contain an Upstream-Status entry
```

priv-kweihmann, 2022-07-06:
Found it - stupid mistake - will do a release with a fix later that day - thank you for reporting (still wondering how that evaded unit testing)

commit 28d4bebc (2022-07-06): Fix upstream Status false positive

by passing the flags by key to the Regex helper

Closes #305