priv-kweihmann/oelint-adv #316: SRC_URI check should not be valid for packagegroups/images

yudjinn, 2022-09-22:
I expect that either the SRC_URI mandatory var is not set for packagegroups 

Another possible resolution is allow # nooelint tags to be inherited in some way so I can make a bbclass that ignores this rule

priv-kweihmann, 2022-09-22:
What particular checks are you referring to?
Do you have a trimmed down example of what is not as expected?

SRC_URI in a bbclass that is inherited in an image/packagegroup already sounds fishy to me

priv-kweihmann, 2022-09-28:
Sorry I think I misread your question on the phone...

As you can see here https://github.com/priv-kweihmann/oelint-adv/blob/92ce35b13351b3728117ea76a3ca15e039915cbd/oelint_adv/rule_base/rule_vars_mandatory_exists.py#L31 the oelint.var.mandatoryvar for SRC_URI is already properly handled for packagegroups and images.

It might be that you're not inheriting packagegroup and/or image classes directly which makes it hard for the parser to guess is the setting is applicable or not.
For images there is a feature at https://github.com/priv-kweihmann/oelint-parser/blob/cea1e82a659390215f437e2807c6f16f52b27570/oelint_parser/data/const-default.json#L147 to add more class that represent an image (see also https://github.com/priv-kweihmann/oelint-parser#working-with-constants).

yudjinn, 2022-09-29:
I can provide a sanitized example tomorrow, but I am inheriting packagegroups and it's still erroring on srcuri as a mandatory.

priv-kweihmann, 2023-01-19:
@yudjinn can you provide an exmaple that still shows the described issue with the latest release version

yudjinn, 2023-01-19:
I will update and see if I still see the issue.

priv-kweihmann, 2023-11-01:
Closing this stale issue - feel free to reopen when new info was added

closed 2023-11-01 by priv-kweihmann