priv-kweihmann/oelint-adv #375: Sort in Alphabetical order does not take parenthesis into account

PercevalSA, 2023-03-02:
Hello, I have the following code

```

RDEPENDS_${PN} += "\
    bash \
    libFormatConversion.so()(64bit) \
    libMvCameraControl.so()(64bit) \
    libMVRender.so()(64bit) \
    libstdc++ \
"
```
Which triggers the following warning `'RDEPENDS' entries should be ordered alphabetically`.

I suspect oelint not to take the part between parenthesis into account.

priv-kweihmann, 2023-03-02:
This is something different - I guess the reason for this report is the implementation of python’s `sorted`.
If you inspect the results it reports
```
['bash', 'libFormatConversion.so', 'libMvCameraControl.so', 'libMVRender.so', 'libstdc++'] # original
['bash', 'libFormatConversion.so', 'libMVRender.so', 'libMvCameraControl.so', 'libstdc++'] # sorted
```
for the given snippet.

To quote the documentation
```
In Python, sorting a list alphabetically is as easy as passing a list of strings to the sorted() method. 
Strings are sorted in alphabetical order based on their first letter (A-Z). 
However, words that start with uppercase letters come before words that start with lowercase letters. 
(Note: This means “Zebra” will be listed before “apple” because “Zebra” starts with a capital letter.)
```

so the report is actually correct, even though the reason is not very obvious I admit that

PercevalSA, 2023-03-02:
Thank your very much for this detailed answer :)

Could you add `key=str.lower` to the `sort` function in order to have case insensitive rule?

priv-kweihmann, 2023-03-03:
Will have a look into that soon

commit 681817dd (2023-03-05): dependsordered: check case insensitive

this works around an oddity of python’s sorted, which
groups uppercase before lowercase.
The rule itself should just enforce an ordering that improves
readability, so exact ordering is not a matter here.
Turn the sorting to case insensitive and add a test case for that

Closes #375
closed 2023-03-05 by priv-kweihmann