priv-kweihmann/oelint-adv #376: requirenotfound does not work accross layers

PercevalSA, 2023-03-02:
Hello,

I have multiple layers and in my project. I'm using "require" in a recipe to include a file from another layer. The build is working perfectly, but oelint says it cannot found the required file as it is not in the same tree or something.

The architecture of my project is:
```
|-layer1
    |-recipes-to-include
        |-myinclude.inc
|-layer2
    |-recipes-image
        |-myimage.bb
```
my image includes `myinclude.inc`

I don't know if my practice is okay. If yes, could we add a search in all folder for this rule?

Thank you

priv-kweihmann, 2023-03-02:
this is intentional - in my opinion layers should be self containing, so cross layer requires/includes are a source of constant potential for breakage.
Hence I would say the warning is justified.
From the linter’s perspective the world ends at the layer boundaries.

What I normally do for these issues is to use the inline `# nooelint: oelint.file.requirenotfound - <comment explaining why this is still okay>` comments - still I think relying on potentially missing includes that result from expecting a more or less hard coded path from another layer is bad practice and should be avoided if possible.
For the remaining things that cannot be done without it, the inline suppression is a welcome tool

closed 2023-03-02 by priv-kweihmann

PercevalSA, 2023-03-02:
I see, I will clean my project then ;)
Thank you