priv-kweihmann/oelint-adv #417: rulefile: enabling single sub-rule enables all other sub-rules of that group

jankraing, 2023-07-24:
E.g. loading the following rulefile
```
{
  "oelint.var.mandatoryvar.DESCRIPTION": "error"
}
```

results in all `mandatoryvar` sub-rules being enabled:

```
Loaded rules:
	oelint.var.mandatoryvar
	oelint.var.mandatoryvar.DESCRIPTION
	oelint.var.mandatoryvar.HOMEPAGE
	oelint.var.mandatoryvar.LICENSE
	oelint.var.mandatoryvar.SRC_URI
	oelint.var.mandatoryvar.SUMMARY
```

Expected behavior is that only the `oelint.var.mandatoryvar.DESCRIPTION` rule is being loaded.

priv-kweihmann, 2023-07-24:
It's just the printout that is listing the rules, results are only reported for sub-ids that are actually part of the rulefile config

priv-kweihmann, 2023-07-24:
Minor inconvenience, will try to fix that later

PR #419 (2023-07-24): rulefile: print only applicable rules
Closes #417

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check even if n.a.)
closed 2023-07-24 by priv-kweihmann