priv-kweihmann/oelint-adv #421: `--fix` applies suppressed rules

alub, 2023-08-09:
Hi,

The suppressed rules listed in the configuration are rightly removed from the output (although it seems they are still executed, is there a reason for that?), but running `--fix` edits files with all the rules, not just the ones that remain. That is especially bothersome when using `oelint-adv` for the first time on a codebase as the resulting diff is way too big for review.

I’d be happy to contribute a fix, but I’d first like to understand first why suppressed rules at executed at all.

priv-kweihmann, 2023-08-11:
That has to be something else.

`fix()` is only called here https://github.com/priv-kweihmann/oelint-adv/blob/2c7341d0eb701983901d3931de2ff063c3176655/oelint_adv/__main__.py#L316 and that function is only run from here https://github.com/priv-kweihmann/oelint-adv/blob/2c7341d0eb701983901d3931de2ff063c3176655/oelint_adv/__main__.py#L348.

Filtering is done since 3.22.4 in the expected way, as `rules` is already prefiltered according to the passed `rulefile`.

So with the latest version, I can't reproduce the issue

alub, 2023-08-11:
OK, thanks for your answer, I get it now: I don’t use a `rulefile`, I just listed rules with `suppress` in `.oelint.cfg`.

I can easily switch to a rulefile, but what’s the intended use for `suppress`?

priv-kweihmann, 2023-08-11:
That was initially the idea to suppress findings, the rulefile came - but now I get it, will have a look into the suppression or deprecate the argument

commit f52bc05d (2023-08-12): main: rework suppression handling

instead of just removing things in the output, handle
suppression the same way as rule files and merge them into
a definitive list of what should be run.
This will also prevent that suppressed rules actually will run
the fix methods but don't show output.

Closes #421
PR #425 (2023-08-12): main: rework suppression handling
instead of just removing things in the output, handle suppression the same way as rule files and merge them into a definitive list of what should be run.
This will also prevent that suppressed rules actually will run the fix methods but don't show output.

Closes #421

# Pull request checklist

## Bugfix

- [ ] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check even if n [... cut]
commit 70e20956 (2023-08-12): main: rework suppression handling

instead of just removing things in the output, handle
suppression the same way as rule files and merge them into
a definitive list of what should be run.
This will also prevent that suppressed rules actually will run
the fix methods but don't show output.

Closes #421
closed 2023-08-12 by priv-kweihmann