priv-kweihmann/oelint-adv #436: rulefile: rule groups can only be enabled/disabled as a whole

jankraing, 2023-09-15:
Including the following section in the rulefile.json

```
  "oelint.var.mandatoryvar": "error",
  "oelint.var.mandatoryvar.DESCRIPTION": "error",
  "oelint.var.mandatoryvar.LICENSE": "error",
  "oelint.var.mandatoryvar.SRC_URI": "error",
```

enables all mandatoryvar subrules, including SUMMARY and HOMEPAGE.

Leaving out ` "oelint.var.mandatoryvar": "error",` results in no errors even for DESCRIPTION, LICENSE and SRC_URI.

jankraing, 2023-10-11:
After looking into it, I'm really not sure how to solve this or if you even want to change this behavior.
Maybe a "none" severity level could be added to ignore/exclude (sub-)rules like so:

```
 if _severity == 'none':
            return []
```

For now I will change severity in the rulefile to "none" and ignore issues with "none" severity further down in the pipeline.

priv-kweihmann, 2023-10-11:
Yup, I was thinking about how to solve this as well for quite some time now - your proposal with 'none'/'null' sounds like a way to make it work - will look into it, if work allows that this week

commit 2786f9aa (2023-10-12): rulefile: enabled use of subids only

In the latest rework an issues slipped in that
allowed only entire groups (or ID) to be
enabled/disabled, mind the subIDs as well when
loading the rules.
As they are filtered out afterwards it should be fine.
This applies of course only to the rule itself,
thus we don't run into conflicts with the fix method call

Closes #436
PR #443 (2023-10-12): rulefile: enabled use of subids only
In the latest rework an issues slipped in that
allowed only entire groups (or ID) to be
enabled/disabled, mind the subIDs as well when
loading the rules.
As they are filtered out afterwards it should be fine. This applies of course only to the rule itself,
thus we don't run into conflicts with the fix method call

Closes #436

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] [... cut]
commit 06dbba27 (2023-10-12): rulefile: enabled use of subids only

In the latest rework an issues slipped in that
allowed only entire groups (or ID) to be
enabled/disabled, mind the subIDs as well when
loading the rules.
As they are filtered out afterwards it should be fine.
This applies of course only to the rule itself,
thus we don't run into conflicts with the fix method call

Closes #436
closed 2023-10-12 by priv-kweihmann