priv-kweihmann/oelint-adv #461: [Question] Behaviour of rules when parsing .bbappend files with require statements

MQueiros, 2023-12-23:
Hi!

I have a bbappend file that roughly looks like:
```
require path/to/somerequire.inc
require path/to/somerequire2.inc

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI += "\
    file://something.patch \
```

If I execute some rule for this recipe, oelint just prints "Parsing" and skips the validation:

```bash
$ oelint-adv --relpaths path/to/somefile_%.bbappend --customrules /path/to/container/oelint/rules --rulefile /path/to/container/oelint/rules/rulefile_project_patching.json --constantmods /path/to/container/oelint/rules/project-constants.json
Loaded rules:
        rule.patch.detect
Parsing path/to/somefile_%.bbappend
Parsing path/to/somerequire.inc
Parsing path/to/somerequire2.inc
```

However, the same works fine if I rename the file to .bb extension. Is there any particular reason for this beaviour?

MQueiros, 2023-12-23:
No, the same rule executes fine for other `bbappend` files. It's just that when it executes for a file that starts with `require` lines, it parses those required files, but then it doesn't actually run the `check` code.
I'll try to check a snippet later when I come back to the office

priv-kweihmann, 2023-12-23:
```
require path/to/somerequire.inc
require path/to/somerequire2.inc

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI += "\
    file://something.patch \
```

gives me no results as well, but that's due to the corrupted syntax

```
require path/to/somerequire.inc
require path/to/somerequire2.inc

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI += "\
    file://something.patch \
"
```

works as expected. So it has actually nothing do to with the require statements, but with a syntax error.
Will try to fix that later in the parser

MQueiros, 2023-12-23:
That's actually because I tried to create a minimal snippet and forgot the quotes at the end. I'll add a better example later.