priv-kweihmann/oelint-adv #467: oelint output when parsing bbappend files with `require` lines

MQueiros, 2024-01-04:
Consider the case when a `path/to/somefile.bbappend` file roughly looks like:
```
require path/to/somerequire.inc
require path/to/somerequire2.inc

FILESEXTRAPATHS:prepend := "${THISDIR}/files:"

SRC_URI += "\
    file://something.patch \
"
```

And also that `path/to/somerequire.inc` causes an oelint error.

If I run the rule multiple times, sometimes I get:

```
path/to/somerequire.inc:2:error:
```

Other times I get:

```
path/to/somefile_%.bbappend:5:error:
```

If the `onlyappend` option is used, only the second case will throw errors.
Also, a more deterministic output would be preferable for making it possible to have post-processing tooling catching the errors.

priv-kweihmann, 2024-01-04:
I tried it now for almost a 100 times with different settings, every time I get the very same results.
Judging from the related discussion you are using custom rule classes, maybe it might be worth sharing a redacted version of that one.
Or post an updated example of what's wrong

MQueiros, 2024-01-04:
Ah, that's true. My bad. I was calling `self.finding` with the wrong filename in some cases. Thanks! :)

closed 2024-01-04 by MQueiros