priv-kweihmann/oelint-adv #468: Misspell warning in SRCREV when using SRCREV_FORMAT syntax

fbertux, 2024-01-04:
Hi!

There is an issue, seems related to version 2.13.0, that throws an oelint.vars.mispell warning if a recipe uses an override and also has SRCREV_FORMAT.

To reproduce the issue, we can use the following recipe:
```
SUMMARY = "ZNC, an advanced IRC bouncer"
DESCRIPTION = "ZNC, an advanced IRC bouncer"
HOMEPAGE = "foo"
LICENSE = "GPL-2.0-only"
LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"

# Line that introduce the issue
FOO:qemux86 = "1"

SRC_URI = "git://github.com/znc/znc.git;name=znc;branch=master;protocol=https \
                  git://github.com/jimloco/Csocket.git;destsuffix=git/third_party/Csocket;name=Csocket;branch=master;protocol=https \
"
SRCREV_znc = "bf253640d33d03331310778e001fb6f5aba2989e"
SRCREV_Csocket = "e8d9e0bb248c521c2c7fa01e1c6a116d929c41b4"

SRCREV_FORMAT = "znc_Csocket"
```

and run the following command:
```
oelint-adv znc_1.8.2.bb
```

I see the following message:
```
znc_1.8.2.bb:13:warning:oelint.vars.mispell:'SRCREV_' is unknown, maybe you meant 'SRCREV'
```

But removing line `FOO:qemux86 = "1"` the oelint.vars.mispell warning doesn't appear.

I ran the sequence above using oelint-parser version 2.12.3, using the command `pip3 install --force-reinstall -v "oelint-parser==2.12.3"`, and didn't see the oelint.vars.mispell warning. Changing to version 2.13.0, `pip3 install --force-reinstall -v "oelint-parser==2.13.0"`, I can reproduce the issue. The issue persists in the current 2.13.3 release.

Is this a known issue? Do you need additional information? 

Thanks!

fbertux, 2024-01-05:
Hi!
I tried it today with version 3.26.8, and I still see the same problem. If I add `FOO:qemux86 = "1"` to line 7 in [this](https://git.openembedded.org/meta-openembedded/plain/meta-networking/recipes-irc/znc/znc_1.8.2.bb) recipe, I get the warning `warning:oelint.vars.mispell:'SRCREV_' is unknown, maybe you meant 'SRCREV'`

I checked versions with this pip command:
```
$ pip3 install oelint_adv --upgrade
Requirement already up-to-date: oelint_adv in /home/fberton/.local/lib/python3.8/site-packages (3.26.8)
Requirement already satisfied, skipping upgrade: anytree~=2.12 in /home/fberton/.local/lib/python3.8/site-packages (from oelint_adv) (2.12.0)
Requirement already satisfied, skipping upgrade: urllib3~=2.1 in /home/fberton/.local/lib/python3.8/site-packages (from oelint_adv) (2.1.0)
Requirement already satisfied, skipping upgrade: wheel~=0.42 in /home/fberton/.local/lib/python3.8/site-packages (from oelint_adv) (0.42.0)
Requirement already satisfied, skipping upgrade: colorama~=0.4 in /home/fberton/.local/lib/python3.8/site-packages (from oelint_adv) (0.4.6)
Requirement already satisfied, skipping upgrade: oelint-parser~=2.13 in /home/fberton/.local/lib/python3.8/site-packages (from oelint_adv) (2.13.3)
Requirement already satisfied, skipping upgrade: six in /home/fberton/.local/lib/python3.8/site-packages (from anytree~=2.12->oelint_adv) (1.16.0)
Requirement already satisfied, skipping upgrade: regex==2023.12.25 in /home/fberton/.local/lib/python3.8/site-packages (from oelint-parser~=2.13->oelint_adv) (2023.12.25)

$ oelint-adv --version
oelint-adv 3.26.8
```

Am I missing something?

priv-kweihmann, 2024-01-05:
@fbertux I can't reproduce your issue - as you can see this test code by you is part of the test cases now and all of them pass without any issue.
Version matrix of the tools look good to me.

Could you come up with another way that issue occurs to you, please

fbertux, 2024-01-05:
The issue happens for all recipes that have `SRCREV_FORMAT`  `SRCREV_foo` syntax and have an override before the `SRCREV_foo`. If the override is after `SRCREV_foo` the linter doesn't show the oelint.vars.mispell.

To me, if I change the [znc](https://git.openembedded.org/meta-openembedded/plain/meta-networking/recipes-irc/znc/znc_1.8.2.bb) recipe by adding an override before line `SRCREV_znc` I can reproduce the error. The `FOO:qemux86 = "1"` will trigger the error, also changing the `DEPENDS` line to `DEPENDS:class-native = "openssl zlib icu"` triggers the error. But If I added the  `FOO:qemux86 = "1"` after the `SRCREV_foo`, e.g. bellow `SRCREV_FORMAT`, the error does not appear.

priv-kweihmann, 2024-01-06:
Run in python:3.8 docker container.
```
oelint-adv --relpaths --quiet meta-networking/recipes-irc/znc/znc_1.8.2.bb 
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:info:oelint.var.suggestedvar.CVE_PRODUCT:Variable 'CVE_PRODUCT' should be set
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:info:oelint.var.suggestedvar.BUGTRACKER:Variable 'BUGTRACKER' should be set
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:info:oelint.var.suggestedvar.AUTHOR:Variable 'AUTHOR' should be set
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:error:oelint.var.mandatoryvar.DESCRIPTION:Variable 'DESCRIPTION' should be set
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:info:oelint.var.bbclassextend:BBCLASSEXTEND should be set if possible
meta-networking/recipes-irc/znc/znc_1.8.2.bb:1:error:oelint.var.mandatoryvar.HOMEPAGE:Variable 'HOMEPAGE' should be set
meta-networking/recipes-irc/znc/znc_1.8.2.bb:6:warning:oelint.vars.dependsordered:'DEPENDS' entries should be ordered alphabetically
meta-networking/recipes-irc/znc/znc_1.8.2.bb:8:info:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 0 set, 4 desirable
meta-networking/recipes-irc/znc/znc_1.8.2.bb:9:info:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 11 set, 4 desirable
```

I suspect something is off with your local python site-packages dir.

Could you please try to remove the linter and install it entirely fresh

fbertux, 2024-01-07:
Did you change the recipe by adding an override before `SRCREV_znc` line?

I did these steps using Docker:
```
$ docker run -it --rm -v "$PWD":/opt -w /opt python:3.8 /bin/bash
root@52f5c09d2860:/opt# pip3 install oelint_adv
Collecting oelint_adv
  Downloading oelint_adv-3.26.10-py2.py3-none-any.whl (86 kB)
     --- 86.6/86.6 kB 1.8 MB/s eta 0:00:00
Collecting anytree~=2.12
  Downloading anytree-2.12.1-py3-none-any.whl (44 kB)
     --- 44.9/44.9 kB 2.7 MB/s eta 0:00:00
Collecting colorama~=0.4
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Requirement already satisfied: wheel~=0.42 in /usr/local/lib/python3.8/site-packages (from oelint_adv) (0.42.0)
Collecting oelint-parser~=2.13
  Downloading oelint_parser-2.13.11-py2.py3-none-any.whl (32 kB)
Collecting urllib3~=2.1
  Downloading urllib3-2.1.0-py3-none-any.whl (104 kB)
     --- 104.6/104.6 kB 4.0 MB/s eta 0:00:00
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting regex==2023.12.25
  Downloading regex-2023.12.25-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (777 kB)
     --- 777.0/777.0 kB 5.4 MB/s eta 0:00:00
Installing collected packages: urllib3, six, regex, colorama, oelint-parser, anytree, oelint_adv
Successfully installed anytree-2.12.1 colorama-0.4.6 oelint-parser-2.13.11 oelint_adv-3.26.10 regex-2023.12.25 six-1.16.0 urllib3-2.1.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 23.0.1 -> 23.3.2
[notice] To update, run: pip install --upgrade pip
root@52f5c09d2860:/opt# oelint-adv --relpaths --quiet znc_1.8.2.bb
znc_1.8.2.bb:1:error:oelint.var.mandatoryvar.DESCRIPTION:Variable 'DESCRIPTION' should be set
znc_1.8.2.bb:1:info:oelint.var.bbclassextend:BBCLASSEXTEND should be set if possible
znc_1.8.2.bb:1:info:oelint.var.suggestedvar.CVE_PRODUCT:Variable 'CVE_PRODUCT' should be set
znc_1.8.2.bb:1:info:oelint.var.suggestedvar.AUTHOR:Variable 'AUTHOR' should be set
znc_1.8.2.bb:1:info:oelint.var.suggestedvar.BUGTRACKER:Variable 'BUGTRACKER' should be set
znc_1.8.2.bb:1:error:oelint.var.mandatoryvar.HOMEPAGE:Variable 'HOMEPAGE' should be set
znc_1.8.2.bb:6:warning:oelint.vars.dependsordered:'DEPENDS' entries should be ordered alphabetically
znc_1.8.2.bb:8:info:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 0 set, 4 desirable
znc_1.8.2.bb:9:info:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 11 set, 4 desirable
znc_1.8.2.bb:11:warning:oelint.vars.mispell:'SRCREV_' is unknown, maybe you meant 'SRCREV'
```

But I changed the line 6 to `DEPENDS:class-native = "openssl zlib icu"`, the error only happens when adding a variable with override before `SRCREV_znc` line.

This recipe will show the issue:
```
SUMMARY = "ZNC, an advanced IRC bouncer"
SECTION = "net"
LICENSE = "GPL-2.0-only"
LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"

DEPENDS:class-native = "openssl zlib icu"

SRC_URI = "git://github.com/znc/znc.git;name=znc;branch=master;protocol=https \
           git://github.com/jimloco/Csocket.git;destsuffix=git/third_party/Csocket;name=Csocket;branch=master;protocol=https \
          "
SRCREV_znc = "bf253640d33d03331310778e001fb6f5aba2989e"
SRCREV_Csocket = "e8d9e0bb248c521c2c7fa01e1c6a116d929c41b4"
```

priv-kweihmann, 2024-01-07:
@fbertux I finally found it, it is indeed an interesting corner case - the filename is important here - wrote a patch and added a test case, to avoid future regressions.
Thank you for finding that one

commit 1d726b12 (2024-01-04): misspell-rule: also use named resources

from SRC_URI to identify variables names.
In before this issue was also hidden due to too high confidence
setting in the best match function.

Lower the confidence to 0.5 and add named resources for
at leats SRCREV_{x} pattern as valid and known variables.

Closes #468
PR #469 (2024-01-04): misspell-rule: also use named resources
from SRC_URI to identify variables names.
In before this issue was also hidden due to too high confidence setting in the best match function.

Lower the confidence to 0.5 and add named resources for at leats SRCREV_{x} pattern as valid and known variables.

Closes #468

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check ev [... cut]
commit c76c68f0 (2024-01-04): misspell-rule: also use named resources

from SRC_URI to identify variables names.
In before this issue was also hidden due to too high confidence
setting in the best match function.

Lower the confidence to 0.5 and add named resources for
at leats SRCREV_{x} pattern as valid and known variables.

Closes #468
PR #478 (2024-01-07): misspell: bail out earlier for known vars
so we could catch valid vars like expanded statements of SRCREV_${PN} from the extra resources list.
The error surfaced only if a named resource has the very same name as one of the valid package names.

Closes #468

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check even if n.a.)
closed 2024-01-07 by priv-kweihmann