priv-kweihmann/oelint-adv #481: AUTHOR is deprecated

jo-so-nx, 2024-01-08:
According to [this](https://lists.openembedded.org/g/openembedded-devel/message/108009) and [this](https://lists.openembedded.org/g/openembedded-devel/message/108108) message the AUTHOR field is no longer in use for package generation and it's usage is discouraged.

priv-kweihmann, 2024-01-08:
Still this is pretty much valid for older releases, thus I tend to keep it.
I am currently drawing out a plan for version 4.x, that will support yocto specific release tweaks.

So we will have to put that one on the backlog for now

priv-kweihmann, 2024-01-08:
as a temporary workaround you could do the following

```shell
oelint-adv --quiet --print-rulefile > rules.json
sed '/oelint.var.suggestedvar.AUTHOR/d' rules.json
```

and in the following always run the tool with `oelint-dev --rulefile=rules.json`, so the rule would be not presented for this individual setup.

Also see https://github.com/priv-kweihmann/oelint-adv#configuration-file to further automate it

closed 2024-01-17 by priv-kweihmann