priv-kweihmann/oelint-adv #482: Bogus warnings since 3.26.8

jo-so, 2024-01-08:
With version 3.26.8 oelint-adv give strange warnings I don't get with 3.26.7. Bitbake compiles all files fine.

```text
meta/recipes-core/packagegroups/packagegroup-container.bbappend:22:warning:oelint.vars.mispell:'MULTICONFIG_HOST_BUILD' is unknown, maybe you meant 'CONFIG_AUDIT_FILE'
meta/recipes-core/images/install.bb:46:warning:oelint.vars.mispell:'EXTRA_USERS_PARAMS' is unknown, maybe you meant 'USERADD_PARAM'
meta/xpq-apps-common.inc:11:warning:oelint.vars.mispell:'LIBCOMMON_INSTALL_DIR' is unknown, maybe you meant 'LLVM_INSTALL_DIR'
meta/recipes-xpq/xpq-gui/xpq-gui.inc:43:warning:oelint.vars.mispell:'REQUIRED_MACHINE_FEATURES' is unknown, maybe you meant 'MACHINE_FEATURES'
meta/recipes-xpq/xpq-gui/xpq-gui.inc:48:warning:oelint.vars.mispell:'EXTRA_QMAKEVARS_PRE' is unknown, maybe you meant 'EXTRA_OEMAKE'
meta/xpq-apps-common.inc:6:warning:oelint.vars.mispell:'XPQ_QT_APP_INSTALL_DIR' is unknown, maybe you meant 'LLVM_INSTALL_DIR'
meta/recipes-xpq/bt-client2/bt-client2.inc:35:warning:oelint.vars.mispell:'REQUIRED_MACHINE_FEATURES' is unknown, maybe you meant 'MACHINE_FEATURES'
meta/recipes-bsp/u-boot/u-boot_2021.04.bb:66:warning:oelint.vars.mispell:'UBOOT_NAME' is unknown, maybe you meant 'UBOOT_MACHINE'
meta/recipes-bsp/mkimage/mkimage_git.bb:22:warning:oelint.vars.mispell:'BOOT_TOOLS' is unknown, maybe you meant 'HOSTTOOLS'
meta/recipes-kernel/linux/linux_5.10.bb:40:warning:oelint.vars.mispell:'KCONFIG_MODE' is unknown, maybe you meant 'CONFIG_SITE'
meta/recipes-kernel/linux/linux_5.10.bb:69:warning:oelint.vars.mispell:'KBUILD_DEFCONFIG' is unknown, maybe you meant 'BUILD_PREFIX'
meta/recipes-kernel/linux/linux_5.10.bb:116:warning:oelint.vars.mispell:'KERNEL_VERSION_SANITY_SKIP' is unknown, maybe you meant 'KERNEL_VERSION_NAME'
```

priv-kweihmann, 2024-01-09:
The warnings are not bogus, simply these variables have never been added to the variables DB in before.
A prior bug shadowed that unfortunately.

- added all these vars to the variables DB
- Reverted back to the original "confidence" setting, so it's less likely to report that much noise

closed 2024-01-09 by priv-kweihmann

jo-so-nx, 2024-01-09:
Thank you for the very quick response. The new release 3.26.13 solves the problem.

PR #483 (2024-01-09): misspell: make confidence configurable
but go back to 0.8 as a default as too many users were complaining about too many new warnings.
Value can now be controlled by env var OELINT_MISSPELL_CONFIDENCE

Closes #482

# Pull request checklist

## Bugfix

- [ ] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check even if n.a.)
commit 7e9d7ec6 (2024-01-09): misspell: make confidence configurable

but go back to 0.8 as a default as too many users were complaining
about too many new warnings.
Value can now be controlled by env var OELINT_MISSPELL_CONFIDENCE

Closes #482