priv-kweihmann/oelint-adv #489: Release option in config file

offa, 2024-01-19:
The `--release` option is a nice feature :+1:, but as far as I have tested it's not configurable though `.oelint.cfg` yet.

It would be great to enable it in the .cfg file so different tools can use same options.

priv-kweihmann, 2024-01-19:
Damn, you're right - will fix that in the next couple of days

priv-kweihmann, 2024-01-19:
it actually works right out of the box e.g.

```ini
[oelint]
release=kirkstone
```

works for me

priv-kweihmann, 2024-01-19:
Works for me

```ini
[oelint]
release=dunfell
```

does configure dunfell release.
Added an example to the README - feel free to close it if you think that is enough

offa, 2024-01-20:
Thanks for verification, I'll check this again.

offa, 2024-01-22:
Issue on my end, sorry.

closed 2024-01-22 by offa