priv-kweihmann/oelint-adv #524: [feature] Python port

Freed-Wu, 2024-03-01:
Can it provide a python port? Such as:

`test.py`:

```python
#!/usr/bin/env python
from oelint_adv import check, format_result
with open("XXX.bb") as f:
    text = f.read()
result = check(text)
print(format_result(result))
```

```sh
$ ./test.py
/disk/meta-some/cppcheck-native/cppcheck.inc:26:error:oelint.task.nomkdir:'mkdir' shall not be used in do_install. Use 'install'
```

priv-kweihmann, 2024-03-01:
Did I get that right, you want to use oelint-adv as part of a library?
If so, yes that could be done

Freed-Wu, 2024-03-01:
Yes, I want :smile:

closed 2024-03-23 by priv-kweihmann

Freed-Wu, 2024-04-08:
Current python port is:

```python
for issue in run(create_lib_arguments([path])):
```

Can we have an optional document content?

When we

```python
for issue in run(create_lib_arguments([path], [content])):
```

It will ignore the file content of `path`, and use `content` to replace?

It can be helpful for the file content of path is modified, however, the
modification only exist in buffer of editor, have not be written to hard disk.

priv-kweihmann, 2024-04-08:
Sure, I take that into the backlog

commit 15359fcd (2024-03-23): oelint: allow usage as a library

by moving the relevant parts into a new file called core

Closes #524
commit ccda40d6 (2024-03-23): oelint: allow usage as a library

by moving the relevant parts into a new file called core

Closes #524