priv-kweihmann/oelint-adv #540: Allow suppression of file-wide requirements

jo-so, 2024-03-19:
I'm having a custom bbclass in another layer which sets the SRC_URI. Hence, I need to define a nooelint which gets flagged as unneeded.

```
# nooelint: oelint.var.mandatoryvar.SRC_URI                                                                                                                
SUMMARY = "..."

inherit my_class
```

oelint-adv (4.4.5) reports:

```
test.bb:1:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.mandatoryvar.SRC_URI' is not needed
```

priv-kweihmann, 2024-03-20:
Out of layer classes always will be a problem, I'm trying to work on a general solution to but like these very specific parameters into the mix - might take a moment

closed 2024-03-23 by priv-kweihmann

jo-so, 2024-03-24:
Unfortunately, your solution only helps with SRC_URI. But I'm having the same issue with:

```conf
# nooelint: oelint.var.mandatoryvar.LICENSE,oelint.var.mandatoryvar.HOMEPAGE,oelint.var.suggestedvar.BUGTRACKER
```

oelint-adv (5.0.0) reports

```
foo.bb:1:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.mandatoryvar.LICENSE' is not needed
foo.bb:1:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.suggestedvar.BUGTRACKER' is not needed
foo.bb:1:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.mandatoryvar.HOMEPAGE' is not needed
```

For example, Rust uses a special syntax with `!` to apply doc comments `///! …` and attributes `#![…]` to the surrounding (not the following) block/file.

jo-so-nx, 2024-03-25:
Unfortunately, the new patch isn't enough for variables like AUTHOR. I tried

```json
    "oelint-suggestedvar": {
        "AUTHOR-exclude-classes": [
            "foocls"
        ]
    },
```

Maybe a mapping `class → [variables]` would better fit. If you see a class in the recipe, mark all variables as defined.

And I'm unsure if all these changes (even through they help for my use case) really allow single file-wide exceptions for one file, i.e. adding a `# nooelint: …` that addresses the whole file and not the following line.

priv-kweihmann, 2024-03-30:
> Unfortunately, the new patch isn't enough for variables like AUTHOR. I tried
>
> ```json
>     "oelint-suggestedvar": {
>         "AUTHOR-exclude-classes": [
>             "foocls"
>         ]
>     },
> ``` [...]

Haven't added the support for `oelint.var.suggestedvar` yet - will add that to the backlog

commit 290202ff (2024-03-23): mandatory-vars: allow fine tuning

by using the new custom constant feature of the parser

Closes #540