priv-kweihmann/oelint-adv #585: `fix` broken for recipes containing the `require` statement

PLangowski, 2024-06-14:
Running `oelint-adv` with the `--fix` option on recipes that contain `require *.inc`  inserts/moves empty lines in an unpredictable way.

Steps to reproduce bug:

1. Create a recipe that requires an `*.inc file`  and introduce some error so that `fix` is executed on the file (I modified `rubygems-activemodel` as an example):
```
require rubygems-base.inc

# SPDX-License-Identifier: MIT
 SUMMARY = "RubyGem: activemodel"
DESCRIPTION = "A toolkit for building modeling frameworks like Active Record"
HOMEPAGE = "https://rubyonrails.org"

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://MIT-LICENSE;md5=48dbb03e765669b2a017264a9ecc9674"

EXTRA_DEPENDS:append = " "
EXTRA_RDEPENDS:append = " "

DEPENDS:class-native += "\
    rubygems-activesupport-native \
"

GEM_INSTALL_FLAGS:append = " "

SRC_URI[md5sum] = "2aa69c3c00d4e3fdd3137f00e11d656e"
SRC_URI[sha256sum] = "f4c838ea76dfca8967e433ac89603342ae20b65dd61366e62f07120a08e1ad72"

GEM_NAME = "activemodel"

inherit rubygems
inherit rubygentest
inherit pkgconfig

RDEPENDS:${PN}:class-target += "\
    rubygems-activesupport \
"

BBCLASSEXTEND = "native"
```
The whitespace before `SUMMARY` will trigger `fix`

Contents of `rubygems-base.inc`:
```
SOMEVAR="Foo"
```
2. Run `oelint` with `--fix`
3. Observe the changes

```
require rubygems-base.inc

SUMMARY = "RubyGem: activemodel"
# SPDX-License-Identifier: MIT
DESCRIPTION = "A toolkit for building modeling frameworks like Active Record"
HOMEPAGE = "https://rubyonrails.org"

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://MIT-LICENSE;md5=48dbb03e765669b2a017264a9ecc9674"

EXTRA_DEPENDS:append = " "
EXTRA_RDEPENDS:append = " "

DEPENDS:class-native += "\
    rubygems-activesupport-native \
"

GEM_INSTALL_FLAGS:append = " "

SRC_URI[md5sum] = "2aa69c3c00d4e3fdd3137f00e11d656e"
SRC_URI[sha256sum] = "f4c838ea76dfca8967e433ac89603342ae20b65dd61366e62f07120a08e1ad72"

GEM_NAME = "activemodel"

inherit rubygems
inherit rubygentest
inherit pkgconfig
RDEPENDS:${PN}:class-target += "\
    rubygems-activesupport \
"
BBCLASSEXTEND = "native"
```

Notice that the whitespace is correctly removed, but that 2 consecutive empty lines surround `GEM_NAME`.  This violates the style rules. The empty lines between the `inherit` statements and variables below are removed.

This can get even worse; Some recipes look like this after the fix:

```
require <file>.inc

SUMMARY = "Foo"
IMAGE_INSTALL:append = " ..."
(...)
```

priv-kweihmann, 2024-06-15:
Added an improvement, which works for my test cases.
Please give it a try and tell me if it fixes your issues too.
If confirmed I will spin up a new release

PLangowski, 2024-06-17:
Your fix works for me, thank you.

closed 2024-06-17 by priv-kweihmann