priv-kweihmann/oelint-adv #597: False positive with inline functions

UVV-gh, 2024-07-05:
Seems like this case:
`SRC_URI:append = "${@bb.utils.contains('EXTRA_IMAGE_FEATURES', 'my-feature', '', ' file://0001-my-patch.patch', d)}"`
is reported as an `append` without space. Any ideas how to fix it properly?

priv-kweihmann, 2024-07-05:
The inline functions are evaluated always by their true branch statement - if you would negated the condition and switch true and false branch statement it should be fine.
It admit it's a limitation of the parser used

priv-kweihmann, 2024-07-05:
The other option is ofc to inject a pseudo space into the true branch statement

UVV-gh, 2024-07-05:
Any way to disable a warning/error when inline function is used?

priv-kweihmann, 2024-07-06:
Inline suppression should work (https://github.com/priv-kweihmann/oelint-adv#inline-suppression).

priv-kweihmann, 2024-09-03:
Support for negative branch check was added with #611 and released with version 6.0.0

closed 2024-09-03 by priv-kweihmann