priv-kweihmann/oelint-adv #603: Support for multiple SRCREVs?

macpijan, 2024-08-01:
Hi,

Changes in this MR:  https://github.com/3mdeb/meta-trenchboot/pull/36/files

Result in:

```
OOPS - That shouldn't happen: list index out of range - ['recipes-extended/packagegroups/packagegroup-tb.bb', 'recipes-bsp/grub/grub-efi_%.bbappend', 'recipes-kernel/linux/linux-tb_6.6.bb', 'recipes-bsp/grub/grub_%.bbappend']
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib64/python3.12/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/home/macpijan/.cache/pre-commit/repom6gr6rcr/py_env-python3/lib/python3.12/site-packages/oelint_adv/core.py", line 151, in group_run
    fixedfiles += r.fix(f, stash)
                  ^^^^^^^^^^^^^^^
  File "/home/macpijan/.cache/pre-commit/repom6gr6rcr/py_env-python3/lib/python3.12/site-packages/oelint_adv/rule_base/rule_var_multilineindent.py", line 72, in fix
    i.Raw = i.Raw.replace(_lines[index + 1], " " * _likeliest_indent + _lines[index + 1].lstrip())
                          ~~~~~~^^^^^^^^^^^
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/macpijan/.cache/pre-commit/repom6gr6rcr/py_env-python3/lib/python3.12/site-packages/oelint_adv/__main__.py", line 107, in main
    issues = run(args)
             ^^^^^^^^^
  File "/home/macpijan/.cache/pre-commit/repom6gr6rcr/py_env-python3/lib/python3.12/site-packages/oelint_adv/core.py", line 385, in run
    issues = flatten(pool.map(partial(group_run, quiet=args.quiet, fix=args.fix,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/multiprocessing/pool.py", line 774, in get
    raise self._value
IndexError: list index out of range
```

I suspect this is related to the `SRCREV_something` syntax, as without this change, it does not fail like this. Is this expected, simply not supported, or I am missing something here? Any advice how to fix this?

priv-kweihmann, 2024-08-02:
It looks more to me like the fix function running on the, now multiline, SRC_URI, is causing this issue.
There were recent changes to that functionality.

 As you are invoking the linter through precommit, you could try it with an older version to see when it stops regressing.

Sadly I won't have the time in the next weeks to look into that more, so going back to an older version might be the only option, till I figured that out, or someone is providing a PR in the meantime.

An alternative might be to NOT run the linter with the --fix switch, but that might reduce the functionality for your use case

macpijan, 2024-08-09:
It works flawlessly with the most recent v5.7.2

closed 2024-08-09 by macpijan