priv-kweihmann/oelint-adv #621: Git Fetcher: usehead parameter implies nobranch parameter

pasweistorz, 2024-10-11:
Hi,

when using the Git fetcher module the `usehead` parameter implies the `nobranch` parameter. Thus, there should be no need to explicitly specify the `nobranch` parameter when using the `usehead` parameter. But if the `branch` parameter is missing, OELint wants the user to specify the `nobranch` parameter although the `usehead` is specified.

Thanks for investigating.

priv-kweihmann, 2024-10-11:
That sounds awfully complicated and requires a lot of knowledge about the internals of the ever-changing fetcher modules.
The linter currently already suggests to add either `branch` or `nobranch` to the git fetcher, so I would back down to the position and say that this is beyond the scope of a generic linter.

I will keep the issue open if someone is interested in implementing that in a manageable and generic way, but likely I will not add a code myself for that.

pasweistorz, 2024-10-11:
Wouldn't it be enough to just add `usehead` to `_required_unless_options['branch']`?

priv-kweihmann, 2024-10-11:
will check on that

commit b64512ef (2024-10-13): oelint.vars.srcurioptions: add usehead implications

as usehead implies nobranch, add that to the branch check.
Furthermore check that branch or nobranch is set
when usehead is used.

Closes #621
closed 2024-10-13 by priv-kweihmann