priv-kweihmann/oelint-adv #626: Require a file from another layer

pasweistorz, 2024-10-11:
Hi,

when a file is required from another layer, OELint triggers a `oelint.file.requirenotfound` finding. In case the required file contains mandatory variables like `SUMMARY`, `DESCRIPTION` or `HOMEPAGE`, OELint triggers also `oelint.var.mandatoryvar.VARIABLE` findings for them. I think if one decides to include a file from another layer and accept such a finding via a `# nooelint` comment variables defined in the included file shouldn't be considered as missing.

By the way, is it really good to forbid includes from other layers via this rule?
- Bitbake accepts them
- They can help to reduce code duplication
- In case the variable from the other layer is renamed one gets a **build time** error which is not that bad

Thanks for investigating.

priv-kweihmann, 2024-10-11:
I get your point with code reduction and reuse, **but** every layer should be self-confined (at least that's my golden rule) so requiring a file from a different layer is always standing on shaky grounds, as any update (or even a checkout in a different revision can lead to unforeseen difficulties).

I'm okay with bbclasses being in another layer, as this is pretty much standard, but cross-layer includes are the worst thing to debug ever.

To me that's a clear code smell, I wouldn't let pass in an of my code reviews, so IMHO the linter is absolutely right.

And to be frank, the reduction in code in that way comes with a huge burden of technical debt, as the other layer can move into any direction (at least that is what the linter has to assume)

See also #614

closed 2024-10-11 by priv-kweihmann

pasweistorz, 2024-10-11:
Ok, I understand your argumentation, but what about the actual point of this issue. If somebody decides to use such a file from another layer than I don't understand why the linter complains about missing variables if they're defined in the included file.

priv-kweihmann, 2024-10-11:
The point is how should the linter know?
Did you pass the needed file via CLI?
If not it's completely unknown to the linter.

I'm fine with people using this kind of setup, but then just feed the linter with needed nooelint tags to ensure somebody knowledgeable is aware of the risks.

To be clear from my definition of the linter, everything outside of the actual layer in question is opaque and about to be not trustworthy in the long run.

With that in mind every cross layer inheritance apart from class code is turning up some potential false positives (which should be a pointer to either rethink the current implementation or let the linter know that the risks are acceptable in this setup)