priv-kweihmann/oelint-adv #635: Problem with constantmods in v6.5.1

TomaszAIR, 2024-10-21:
After latest update I am unable to run oelint-adv with configuration that work correctly on v6.5.0.

Problem:

```
$ oelint-adv --version
oelint-adv 6.5.1
$ oelint-adv recipes-bsp/u-boot/u-boot-ti-staging_%.bbappend 
Traceback (most recent call last):
  File "/opt/pipx/bin/oelint-adv", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/pipx/venvs/oelint-adv/lib/python3.12/site-packages/oelint_adv/__main__.py", line 106, in main
    args = arguments_post(parse_arguments())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/pipx/venvs/oelint-adv/lib/python3.12/site-packages/oelint_adv/core.py", line 364, in arguments_post
    args = Tweaks.tweak_args(args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/pipx/venvs/oelint-adv/lib/python3.12/site-packages/oelint_adv/tweaks.py", line 102, in tweak_args
    args.constantmods.insert(0, extramod)
    ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'insert'
$ cat .oelint.cfg 
[oelint]
quiet = true
suppress = 
    oelint.var.suggestedvar
    oelint.var.bbclassextend
constantmods =
    +.oelint-constants.json
$ cat .oelint-constants.json 
{
    "variables": {
      "known": [
        "UBOOT_ENV",
        "UBOOT_ENV_SUFFIX"
      ]
    }
  }
```

If I remove `.oelint-constants.json` from config, linter can be executed:

```
$ cat .oelint.cfg 
[oelint]
quiet = true
suppress = 
    oelint.var.suggestedvar
    oelint.var.bbclassextend
$ oelint-adv recipes-bsp/u-boot/u-boot-ti-staging_%.bbappend 
/home/ubuntu/meta-tr-a/meta-tr-a/recipes-bsp/u-boot/u-boot-ti-staging_%.bbappend:7:info:oelint.vars.mispell.unknown:'UBOOT_ENV' is unknown. Consider adding it via --constantmods [branch:true]
/home/ubuntu/meta-tr-a/meta-tr-a/recipes-bsp/u-boot/u-boot-ti-staging_%.bbappend:8:warning:oelint.vars.mispell:'UBOOT_ENV_SUFFIX' is unknown, maybe you meant 'UBOOT_SUFFIX' [branch:true]
```

Also, this worked on v6.4.0.

```
$ oelint-adv --version
oelint-adv 6.4.0
$ cat .oelint.cfg 
[oelint]
quiet = true
suppress = 
    oelint.var.suggestedvar
    oelint.var.bbclassextend
constantmods =
    +.oelint-constants.json
(failed reverse-i-search)`cat': ^Ct .oelint.cfg 
$ oelint-adv --version
oelint-adv 6.4.0
$ cat .oelint.cfg 
[oelint]
quiet = true
suppress = 
    oelint.var.suggestedvar
    oelint.var.bbclassextend
constantmods =
    +.oelint-constants.json
$ cat .oelint-constants.json 
{
    "variables": {
      "known": [
        "UBOOT_ENV",
        "UBOOT_ENV_SUFFIX"
      ]
    }
  }
$ oelint-adv recipes-bsp/u-boot/u-boot-ti-staging_%.bbappend
```

1. Is my json file wrongly configured?

priv-kweihmann, 2024-10-21:
Dang, thanks for reporting that. A fix PR is out and a new release should be available later that day

TomaszAIR, 2024-10-21:
@priv-kweihmann thanks for quick response. I will check a new version when it will be available and report back results.

PR #636 (2024-10-21): core/post: deserialize all options before tweaks
as e.g. constantmods is expected to be a list,
but if it's set through the config file, it likely is still a string, hence the need to deserialize before applying the tweaks

Closes #635
closed 2024-10-21 by priv-kweihmann