priv-kweihmann/oelint-adv #639: oelint.task.nopythonprefix falsely reports positive results on simple Python scripts

skycaptain, 2024-10-25:
Simple Python scripts, such as the [sanity check from meta-virtualization](https://github.com/lgirdk/meta-virtualization/blob/master/classes/sanity-meta-virt.bbclass), is falsely flagged with the warning: warning:oelint.task.nopythonprefix:Tasks containing shell code, should not be prefixed with python in function header. The script, however, is undoubtedly valid Python code, not shell code.

priv-kweihmann, 2024-10-26:
```
addhandler virt_bbappend_distrocheck
virt_bbappend_distrocheck[eventmask] = "bb.event.SanityCheck"
python virt_bbappend_distrocheck() {
    skip_check = e.data.getVar('SKIP_META_VIRT_SANITY_CHECK') == "1"
    if 'virtualization' not in e.data.getVar('DISTRO_FEATURES').split() and not skip_check:
        bb.warn("You have included the meta-virtualization layer, but \
                'virtualization' has not been enabled in your DISTRO_FEATURES. Some bbappend files \
                may not take effect. See the meta-virtualization README for details on enabling \
                virtualization support.")
}
```

or any other version where the line continued bb.warn is indented at least one level doesn't throw this warning.
The check uses ast.parse to check for syntax and that one flags the original code as not valid python code.

The only thing I see is that the message is a little misleading (that could be tweaked) but the check is correct IMO

skycaptain, 2024-10-26:
Ah, you're right. I now see the issue. The example is actually invalid Python code, and the problem stems from the multiline string in the wrapping BitBake syntax. So yes, the check is correct if its intention is to verify valid pure-Python code without any BitBake-specific additions. I'm uncertain about the best practice in this case. Should oelint-adv be updated to recognize these BitBake-specific additions, or should this be considered a code smell, encouraging developers to write valid Python code without relying on BitBake syntax?

priv-kweihmann, 2024-10-27:
Here's what I propose:

- tweak the message on the linter's end (explicitly naming "valid" python code as the reason)
- keep labeling the issue as an error

I come to believe it's using a shortcoming of the bitbake parser and not really producing a valid python syntax.
E.g. it would be impossible to move this functionality to a lib based file (I strongly believe this would produce a parser error on bitbake's end).

So I would personally go with the second track: python tasks need to have valid python syntax without any bitbake specific tricks

priv-kweihmann, 2024-10-28:
New message in 6.5.4 should make it more clear

closed 2024-10-28 by priv-kweihmann