priv-kweihmann/oelint-adv #640: oelint.var.multiinclude does not support expanding variables

skycaptain, 2024-10-25:
Consider meta-montavista-cgx as an example of a pattern where an .inc file is included only when a certain variable is set: https://github.com/MontaVista-OpenSourceTechnology/meta-montavista-cgx/blob/4a74d7d1d956a3320e829eaec14e06036fa0307c/recipes-devtools/libunwind/libunwind_1.6.%25.bbappend

When this pattern is used across multiple bbappends, it triggers the following false positive: warning:oelint.var.multiinclude:'${PN}_mvista.inc' is included multiple times

This warning highlights two issues: First, it appears that this check operates at the global distro level instead of examining includes per recipe. Second, variables are not expanded, which would correctly resolve into unique include filenames.

priv-kweihmann, 2024-10-26:
It is true that the include name is not expanded so far, still I don't get how that could result in any sort of warning here.
The linter will group files into logical groups (meaning recipe + matching bbappends), so unless there are multiple bbappends using this scheme that apply to a single recipe, the include statement in question should be parsed only once per group.

Please provide a minimal example on how warning is falsely raised.

skycaptain, 2024-11-01:
While preparing a minimal reproducible example, I tracked down the issue. I had the same filename for an include file in a single layer, once under a regular recipe folder and once as a dynamic layer. Renaming one of the includes solved the issue. However, the question remains: what's the best practice in this case? Having multiple includes with the same name seems to be valid Bitbake code in this case as bitbake -e shows.

priv-kweihmann, 2024-11-02:
Same name shouldn't be a problem, the parser code

```python
def FindLocalOrLayer(self, name: str, localdir: str) -> str:
        """Find file in local dir or in layer

        Arguments:
            name {str} -- filename
            localdir {str} -- path to local dir

        Returns:
            str -- path to found file or None
        """
        if os.path.exists(os.path.join(localdir, name)):
            return os.path.join(localdir, name)
        _curdir = localdir
        while os.path.isdir(_curdir):
            if _curdir == "/":
                break
            _curdir = os.path.dirname(_curdir)
            if os.path.exists(os.path.join(_curdir, "conf/layer.conf")):
                if os.path.exists(os.path.join(_curdir, name)):
                    return os.path.join(_curdir, name)
                else:
                    break
        return ""
 ```
 
 searches in the same dir as the file that had the include, or by full path in the layer.
 So unless there are multiple include files reachable by the same path it should only return a single item.
 
 In terms of includes I think there are two rules of thumb
 
 - keep the include files next to the recipe
 - or include them with a significant enough path prefix (e.g. `include full/path/in/the/layer/file.inc`)
 
Using includes might produce perfectly valid bitbake code, but order and if those files overwrite anything matters from the linter's perspective, while for bitbake itself it doesn't matter at all
 
 Still I don't understand how you got to this error.
 If it's still an issue please, provide an example or close this issue

skycaptain, 2024-11-02:
Sure, with this example you should be able to reproduce the issue when running `pre-commit run --all-files --verbose`: https://github.com/skycaptain/oelint-adv-issue-640

priv-kweihmann, 2024-11-03:
Ok, I finally see what the problem is - fix release is already pending

closed 2024-11-03 by priv-kweihmann

skycaptain, 2024-11-03:
Thanks, works!

commit 036bdff8 (2024-11-03): oelint.var.multiinclude: base rule on paths

and do not just look at the include name, but
at the actual file included, as they could point
to very much different files.
Fix includes a bump of the parser lib to 6.3

Closes #640
PR #652 (2024-11-03): oelint.var.multiinclude: base rule on paths
and do not just look at the include name, but
at the actual file included, as they could point
to very much different files.
Fix includes a bump of the parser lib to 6.3

Closes #640

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior

## New feature

- [ ] A testcase was added to test the behavior
- [ ] ``wiki-creator.py`` was run and a new wiki document was filled with information
- [ ] New functions are documented with docstrings
- [ ] No debug code is left
- [ ] README.md was updated to reflect the changes (check [... cut]
commit 04f531d4 (2024-11-03): oelint.var.multiinclude: base rule on paths

and do not just look at the include name, but
at the actual file included, as they could point
to very much different files.
Fix includes a bump of the parser lib to 6.3

Closes #640