priv-kweihmann/oelint-adv #644: incorrectly concatenating lines after a multiline indent

afenkart, 2024-10-28:
Below is just an example but seen many issues, and can produce if needed. 

- it accidentally merges the last line with the closing quote with the the next line, in this case SSTATE_SKIP_CREATION
- the last item in the FILES list is not indented, instead of 4 spaces as it should, it remains indented with 2 spaces as it was before

I'm using oelint-adv version: 6.5.3

```
--- a/recipes-meta-apps/maintenance-daemon/maintenance-daemon.bb
+++ b/recipes-meta-apps/maintenance-daemon/maintenance-daemon.bb
@@ -27,5 +27,4 @@ do_install_append() {
 FILES_${PN} += "\
     ${sysconfdir}/networkd-dispatcher/routable.d/* \
   ${sysconfdir}/networkd-dispatcher/configured.d/* \
-"
-SSTATE_SKIP_CREATION = "1"
+"SSTATE_SKIP_CREATION = "1"
```

priv-kweihmann, 2024-10-28:
What outcome did you expect here?
To me that sounds more like a parser problem.
If I should take a wild guess, this construct won't be seen as a variable and hence no checks or fixing can be applied

afenkart, 2024-10-28:
Thank you for your feedback. I apologize, but I don't understand your guesses here. Do you mean that SSTATE_SKIP_CREATION or FILES_${PN} is not detected as a variable by the parser?

So, if I understand you correctly, the parser reads the file, creates an internal representation, and then applies the rules to that representation. Finally, the file is recreated based on the modified internal representation. If this is the case, could I just disable all the rules and only check the parsing and recreating process?

I added another example with SRC/S variables:

Just for completeness, this is what I do:

```
oelint-adv --nobackup --fix <recipe.bb>
git diff <recipe.bb>
```
 
Here, the closing quote was merged with the S variable, and the last line's indentation was inconsistent with the previous lines.

```
 SRC_URI = "\
-  file://dual-boot-install-hooks \
-  file://dual-boot-validation.service \
-  file://dual-boot-functions.sh \
-  file://dual-boot \
+    file://dual-boot-install-hooks \
+    file://dual-boot-validation.service \
+    file://dual-boot-functions.sh \
+    file://dual-boot \
   file://sw-versions \
-"
-
-S = "${WORKDIR}"
+"S = "${WORKDIR}"
```

I looked into 'rule_var_multilineindent.py', for no better matching rule coming to mind.

Disabling multilineindent resolves the issue, but it also removes proper indentation, which produces to false positives when comparing different versions of the same recipe.

```
oelint-adv --nobackup --fix --suppress 'oelint.vars.multilineident'
```

commit 9319d479 (2024-10-30): oelint.vars.multilineident: fix check

Closes #644
commit 66e2e46b (2024-10-30): oelint.vars.multilineident: fix check

Closes #644
closed 2024-10-30 by priv-kweihmann