priv-kweihmann/oelint-adv #658: Wrong warning:oelint.vars.dependsordered detection

mvkonovalov, 2024-11-14:
Hello,

for example I have *.inc file with ordered DEPENDS and then I include the file to *.bb which also has  addition DEPENDS ordered.
In the case I get the following issue:
```
warning:oelint.vars.dependsordered:'DEPENDS' entries should be ordered alphabetically
```

It is expected that warning is not triggered as in each files DEPENDS is ordered.

mvkonovalov, 2024-11-14:
If manually merge DEPENDS from *.inc to *.bb and sort resulting DEPENDS then no issue is triggered.

priv-kweihmann, 2024-11-14:
So you answered your own question I guess.

Yes this is expected behavior.
Also iirc using append instead of just adding += to the list should be fine

closed 2024-11-14 by priv-kweihmann

priv-kweihmann, 2024-11-14:
To give a more technical explanation.
It doesn't matter much if the depends are ordered in each file, what really matters is to give everyone a quick overview on a recipe level what this one depends on.
"Hiding" stuff in include files shadows that heavily.

If needed that should be moved to a bbclass (which doesn't produce those warnings).

In conclusion depends in include files are one of the reasons for this rule, so again not a bug for me

mvkonovalov, 2024-11-14:
> So you answered your own question I guess.
> 
> Yes this is expected behavior. Also iirc using append instead of just adding += to the list should be fine

I don't want to manually merge (This was mentioned as proof that issue related to fact that DEPENDS is checked after being appending but not in each files). There are lot of such recipes in poky, U-Boot for example.
Includes are used to share common code, DEPENDS in the case.

mvkonovalov, 2024-11-15:
Thank you. Yesterday I missed  suggesting using append. Now it works nicely.