priv-kweihmann/oelint-adv #677: How to suppress badimagefeature?

jo-so, 2024-12-30:
With

```
# nooelint: oelint.var.badimagefeature.allow-empty-password
# nooelint: oelint.var.badimagefeature.allow-root-login
# nooelint: oelint.var.badimagefeature.empty-root-password
IMAGE_FEATURES:append:dev = " \
    allow-empty-password allow-root-login empty-root-password \
"
```

I get

```
image.bb:20:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.allow' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:21:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.allow' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:22:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.empty' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:23:warning:oelint.var.badimagefeature.empty-root-password:IMAGE_FEATURES contains 'empty-root-password', that is a security risk. Anyone can authenticate as root over serial without a password; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
image.bb:23:info:oelint.var.badimagefeature.allow-root-login:IMAGE_FEATURES contains 'allow-root-login', that is a security risk. Root can login into the system; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
image.bb:23:warning:oelint.var.badimagefeature.allow-empty-password:IMAGE_FEATURES contains 'allow-empty-password', that is a security risk. Users can have accounts without password protection; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
```

And with

```
# nooelint: oelint.var.badimagefeature.allow_empty_password
# nooelint: oelint.var.badimagefeature.allow_root_login
# nooelint: oelint.var.badimagefeature.empty_root_password
IMAGE_FEATURES:append:dev = " \
    allow-empty-password allow-root-login empty-root-password \
"
```

I get

```
image.bb:20:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.allow_empty_password' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:21:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.allow_root_login' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:22:info:oelint.file.inlinesuppress_na:Inline suppression for 'oelint.var.badimagefeature.empty_root_password' is not needed; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.file.inlinesuppress_na.md
image.bb:23:info:oelint.var.badimagefeature.allow-root-login:IMAGE_FEATURES contains 'allow-root-login', that is a security risk. Root can login into the system; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
image.bb:23:warning:oelint.var.badimagefeature.empty-root-password:IMAGE_FEATURES contains 'empty-root-password', that is a security risk. Anyone can authenticate as root over serial without a password; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
image.bb:23:warning:oelint.var.badimagefeature.allow-empty-password:IMAGE_FEATURES contains 'allow-empty-password', that is a security risk. Users can have accounts without password protection; see https://github.com/priv-kweihmann/oelint-adv/blob/6.6.6/docs/wiki/oelint.var.badimagefeature.md [branch:true]
```

If the right way is `<id>` with `_` than the doc page should say it.

priv-kweihmann, 2025-01-03:
3 thought about this

- a fix will be available soon to allow rule suppression that contains `-` in the ID
- multiple rules need to be concatenate by `.` e.g.
  `# nooelint: oelint.var.badimagefeature.allow-empty-password,oelint.var.badimagefeature.allow-root-login,oelint.var.badimagefeature.empty-root-password`
  Having them as separate lines is not supported
 - images do not have packages, hence I don't really get the `:dev` override, or is that a distro/machine setting?

closed 2025-01-03 by priv-kweihmann

jo-so, 2025-01-03:
> 3 thought about this
> 
>     * a fix will be available soon to allow rule suppression that contains `-` in the ID

As always a huge thank-you for your prompt response which allows moving on without have to add intermediate workarounds.

>     * images do not have packages, hence I don't really get the `:dev` override, or is that a distro/machine setting?

Yes, it's a distro override for the development distribution/images.

commit 224808e9 (2025-01-03): core: fix inline suppression

- allow '-' in id
- rewrite filtering to allow multiple items to be mapped
  better to the workflow

Closes #677
PR #678 (2025-01-03): core: fix inline suppression
- allow '-' in id
- rewrite filtering to allow multiple items to be mapped better to the workflow

Closes #677

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior
commit 1505b2d9 (2025-01-03): core: fix inline suppression

- allow '-' in id
- rewrite filtering to allow multiple items to be mapped
  better to the workflow

Closes #677