priv-kweihmann/oelint-adv #681: oelint-adv --suppress oelint.vars.homepageping fails

jo-so, 2025-01-10:
I want to suppress on certain invocations the homepage checking, but this fails?

```
oelint-adv --suppress oelint.vars.homepageping ….bb 
Traceback (most recent call last):
  File "/home/joerg_sommer/kein_backup/oelint-venv/bin/oelint-adv", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/joerg_sommer/kein_backup/oelint-venv/lib/python3.12/site-packages/oelint_adv/__main__.py", line 106, in main
    args = arguments_post(parse_arguments())
                          ^^^^^^^^^^^^^^^^^
  File "/home/joerg_sommer/kein_backup/oelint-venv/lib/python3.12/site-packages/oelint_adv/__main__.py", line 93, in parse_arguments
    return parser.parse_args()  # pragma: no cover
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 1918, in parse_args
    args, argv = self.parse_known_args(args, namespace)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 1928, in parse_known_args
    return self._parse_known_args2(args, namespace, intermixed=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 1957, in _parse_known_args2
    namespace, args = self._parse_known_args(args, namespace, intermixed)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 2198, in _parse_known_args
    start_index = consume_optional(start_index)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/argparse.py", line 2127, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.12/argparse.py", line 2032, in take_action
    action(self, namespace, argument_values, option_string)
  File "/home/joerg_sommer/kein_backup/oelint-venv/lib/python3.12/site-packages/oelint_adv/core.py", line 33, in __call__
    items.extend(RegexRpl.split(r'\s+|\t+|\n+', values.strip('"').strip("'")))
    ^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'extend'

% oelint-adv --version
oelint-adv 6.6.8
```

jo-so, 2025-01-10:
Does the command line option allow an extension of the list from config file, something like `--suppress +oelint.vars.homepageping` or does it only allow replacing the config values?

priv-kweihmann, 2025-01-10:
`--suppress` arguments extend the list not replace it.
So that should be fine.

One issue I found is a mix of a config file and a suppression causing the described behavior.
A fix is on its way

PR #682 (2025-01-10): core: fix config file list handling
if a config file is used and a statement is
used via CLI, the config parts need to be turned into a list

Closes #681

# Pull request checklist

## Bugfix

- [x] A testcase was added to test the behavior
commit af4d7630 (2025-01-10): core: fix config file list handling

if a config file is used and a e.g. statement is
used via CLI, the config parts need to be turned into
a list

Closes #681
closed 2025-01-10 by priv-kweihmann