priv-kweihmann/oelint-adv #695: IMAGE_ROOTFS_MAXSIZE didn't made it in styhead.json and other release specific data

pasweistorz, 2025-02-04:
Hi,

a while ago an issue in oelint-parser due to the unknown variable `IMAGE_ROOTFS_MAXSIZE` was fixed: https://github.com/priv-kweihmann/oelint-parser/issues/263. We use the `meta-sca` layer in a `styhead` based distro. The fix in oelint-parser doesn't seem to help, we still get the `mispell` warnings from oelint-adv when we use the `IMAGE_ROOTFS_MAXSIZE` variable. Our assumption is that the `IMAGE_ROOTFS_MAXSIZE` variable is missing in the file `data/styhead.json`.

By the way, it's not part of `data/walnascar.json`, too.

Can there anything be done about that?

priv-kweihmann, 2025-02-04:
This is in fact a problem in oe-core.

```
$ bitbake-getvar -r core-image-minimal IMAGE_ROOTFS_MAXSIZE
The variable 'IMAGE_ROOTFS_MAXSIZE' is not defined
```

while

```
bitbake-getvar -r core-image-minimal INITRAMFS_MAXSIZE
#
# $INITRAMFS_MAXSIZE
#   set /layers//poky/meta/conf/bitbake.conf:859
#     [_defaultval] "131072"
INITRAMFS_MAXSIZE="131072"
$ bitbake-getvar -r core-image-minimal IMAGE_ROOTFS_SIZE
#
# $IMAGE_ROOTFS_SIZE [3 operations]
#   set /layers//poky/meta/conf/documentation.conf:225
#     [doc] "Defines the size in Kbytes for the generated image."
#   set /layers//poky/meta/conf/bitbake.conf:867
#     [_defaultval] "65536"
#   set? /layers//poky/meta/recipes-core/images/core-image-minimal.bb:11
#     "8192"
# pre-expansion value:
#   "8192"
IMAGE_ROOTFS_SIZE="8192"
```

is fine.
The problem is that there is no default value for `IMAGE_ROOTFS_MAXSIZE`  just some inline code that is trying to use it, if
the variable was defined out of scope of oe-core (or local.conf or whatever).

Best it would be to get a patch into oe-core to have that default value, as the latest version of the linter
greps all variables defined in core automatically (once the patch is part of the branch ofc) at least weekly.

Alternative is a custom constantmod file (e.g. https://github.com/priv-kweihmann/meta-rubygems/blob/master/files/lint/oelint-constants.json) placed in a custom location and added to the command line arguments.

NOTE: `IMAGE_ROOTFS_MAXSIZE` is also not documented in oe-core, so that would be another bug to me :-)

closed 2025-02-04 by priv-kweihmann

priv-kweihmann, 2025-08-25:
Why shouldn't be ideal, it's basically defining an empty value (just like if the variable isn't defined)?
In any other scripting/programming language accessing an undefined variable is a bad thing, so why should bitbake (just because it covers up that fact, by returning an empty value) be exempt from that?

The only sane and maintainable option is to finally have core define default (empty) values for **all** variables (and properly document them with the `[doc]` flag (which is already the case for this variable)).

Another option would be to also take all `[doc]`  flags into account in https://github.com/priv-kweihmann/oelint-data/blob/master/scripts/bitbake-listvars, but as I think patching core is a much better solution, I leave that to the community to work on any patches

pasweistorz, 2026-02-04:
We tried to get a default value for the `IMAGE_ROOTFS_MAXSIZE` variable upstream, but it wasn't accepted: https://lists.openembedded.org/g/openembedded-core/topic/patch_1_2_bitbake_conf_add/111030916.

The `[doc]` flag for that variable was also missing. So we created a patch for that, too: https://lists.openembedded.org/g/openembedded-core/topic/patch_2_2/111030917. That patch was accepted.

Now we have similar problems with the `LICENSE_FLAGS` and `LICENSE_FLAGS_ACCEPTED` variable. Maybe your idea about taking the `[doc]` flag into account would be helpful. I have the feeling that the maintainers of `openembedded-core` and other layers rather agree on patches about missing `[doc]` flags than on patches about missing default values.