priv-kweihmann/oelint-adv #696: Missing known variables FIT_{HASH,SIGN}_ALG in styhead and scarthgap

antgres, 2025-02-05:
Hello,

I get the following error during checking my U-Boot and kernel FIT signature verification process (on top of oe-core commit `184adcbc322f ("dev-manual/building: document the initramfs-framework recipe")` for the `scarthgap` branch:

```
/path/to/my-secure-boot/recipes-kernel/linux/linux-stable_%.bbappend:3:error:oelint.vars.mispell:'FIT_HASH_ALG' is unknown, maybe you meant 'UBOOT_FIT_HASH_ALG' [branch:true]
/path/to/my-secure-boot/recipes-kernel/linux/linux-stable_%.bbappend:4:error:oelint.vars.mispell:'FIT_SIGN_ALG' is unknown, maybe you meant 'UBOOT_FIT_SIGN_ALG' [branch:true]
```

So I checked out the corresponding oe-release.json files

```
$ grep -R -E 'FIT_.*_ALG' oelint_adv/data/*.json
...
oelint_adv/data/kirkstone.json:      "FIT_HASH_ALG",
oelint_adv/data/kirkstone.json:      "FIT_KERNEL_COMP_ALG",
oelint_adv/data/kirkstone.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
oelint_adv/data/kirkstone.json:      "FIT_PAD_ALG",
oelint_adv/data/kirkstone.json:      "FIT_SIGN_ALG",
oelint_adv/data/kirkstone.json:      "UBOOT_FIT_HASH_ALG",
oelint_adv/data/kirkstone.json:      "UBOOT_FIT_SIGN_ALG",
oelint_adv/data/scarthgap.json:      "FIT_KERNEL_COMP_ALG",
oelint_adv/data/scarthgap.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
oelint_adv/data/scarthgap.json:      "UBOOT_FIT_HASH_ALG",
oelint_adv/data/scarthgap.json:      "UBOOT_FIT_SIGN_ALG",
oelint_adv/data/styhead.json:      "FIT_KERNEL_COMP_ALG",
oelint_adv/data/styhead.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
oelint_adv/data/styhead.json:      "UBOOT_FIT_HASH_ALG",
oelint_adv/data/styhead.json:      "UBOOT_FIT_SIGN_ALG",
oelint_adv/data/walnascar.json:      "FIT_HASH_ALG",
oelint_adv/data/walnascar.json:      "FIT_KERNEL_COMP_ALG",
oelint_adv/data/walnascar.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
oelint_adv/data/walnascar.json:      "FIT_PAD_ALG",
oelint_adv/data/walnascar.json:      "FIT_SIGN_ALG",
oelint_adv/data/walnascar.json:      "UBOOT_FIT_HASH_ALG",
oelint_adv/data/walnascar.json:      "UBOOT_FIT_SIGN_ALG",
```

and its missing for scarthgap and styhead, but not for kirkstone and walnascar. I checked out the git log for the scarthgap and styhead json files but nothing interesting.

I also checked the output of `scripts/generate-variables` for the scarthgap and styhead branch:

```
$ scripts/generate-variables scarthgap $PWD/test.scarthgap.json
$ scripts/generate-variables styhead $PWD/test.styhead.json
$ grep -R -E 'FIT_.*_ALG' test.*.json
test.scarthgap.json:      "FIT_HASH_ALG",
test.scarthgap.json:      "FIT_KERNEL_COMP_ALG",
test.scarthgap.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
test.scarthgap.json:      "FIT_PAD_ALG",
test.scarthgap.json:      "FIT_SIGN_ALG",
test.scarthgap.json:      "UBOOT_FIT_HASH_ALG",
test.scarthgap.json:      "UBOOT_FIT_SIGN_ALG",
test.styhead.json:      "FIT_KERNEL_COMP_ALG",
test.styhead.json:      "FIT_KERNEL_COMP_ALG_EXTENSION",
test.styhead.json:      "UBOOT_FIT_HASH_ALG",
test.styhead.json:      "UBOOT_FIT_SIGN_ALG",
```

Interestingly its still missing for styhead which defines FIT_{HASH,SIGN}_ALG in [meta/classes-recipe/kernel-fitimage.bbclass](https://git.openembedded.org/openembedded-core/tree/meta/classes-recipe/kernel-fitimage.bbclass?h=styhead#n59)

Should I open a PR for adding the known variables FIT_{HASH,SIGN}_ALG to scarthgap and styhead json files?

Or maybe its even better to create oe-release.json files automagically for each release and add oe-release.override.json files if something is missing? Maybe via Github Actions? There is a [poky container](https://github.com/crops/poky-container/pkgs/container/poky) for this job.

priv-kweihmann, 2025-02-06:
A PR will not solve the issue in the long run.
What I suspect here is that kernel-fitimage is never inherited in a vanilla poky setup. At least not for the versions in question.
I will check, but not included bbclasses and similar things like used variables without a default value will always be a problem.

And if there are variables missing you are of course free to add them through a ``--constantmod=+filename.json`` argument from CLI.

Will double check when I'm back at a desk in next couple of weeks

antgres, 2025-02-06:
> A PR will not solve the issue in the long run.
> What I suspect here is that kernel-fitimage is never inherited in a vanilla poky setup. At least not for the versions in question.
> I will check, but [...]

Yeah, probably.

> And if there are variables missing you are of course free to add them through a --constantmod=+filename.json argument from CLI.

That is my current strategy. Thought I'd write an issue for it so that it's known.

> Will double check when I'm back at a desk in next couple of weeks

👍

priv-kweihmann, 2025-02-07:
It's exactly as I thought it is -> https://git.yoctoproject.org/poky/commit/?id=3fb215a3af242e2016a146739a69be746ab8b722 fixed the issue on master and was backported to scarthgap ... BUT, there is no patch for styhead upstream.

What I will do is to enable further weekly polling of a new more branches, but a patch for styhead would need to be posted by the community to oe-core

priv-kweihmann, 2025-02-07:
Problem should be gone for scrathgap

antgres, 2025-02-07:
Perfect, thanks 👍

closed 2025-02-07 by antgres