priv-kweihmann/oelint-adv #721: Recipe variables incorrectly suggested as global scope variables

skycaptain, 2025-02-28:
Running oelint-adv 7.2.1 against my metalayer returns the following false positives:

```
meta/some-recipe.bb:10:warning:oelint.vars.bbvars.SUMMARY:Variable 'SUMMARY' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:11:warning:oelint.vars.bbvars.DESCRIPTION:Variable 'DESCRIPTION' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:12:warning:oelint.vars.bbvars.AUTHOR:Variable 'AUTHOR' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:13:warning:oelint.vars.bbvars.HOMEPAGE:Variable 'HOMEPAGE' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:14:warning:oelint.vars.bbvars.BUGTRACKER:Variable 'BUGTRACKER' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:15:warning:oelint.vars.bbvars.SECTION:Variable 'SECTION' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:16:warning:oelint.vars.bbvars.LICENSE:Variable 'LICENSE' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:18:warning:oelint.vars.bbvars.PROVIDES:Variable 'PROVIDES' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:22:warning:oelint.vars.bbvars.PACKAGES:Variable 'PACKAGES' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta/some-recipe.bb:29:warning:oelint.vars.bbvars.EXCLUDE_FROM_WORLD:Variable 'EXCLUDE_FROM_WORLD' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
```

priv-kweihmann, 2025-03-01:
Hi, I tried to reproduce the issue, I don't get any warning.
Are you sure your local environment has everything installed in the latest version?
Recently oelint-data was added as a dependency, which needs to be up to date

skycaptain, 2025-03-02:
I've created a MRE here: https://github.com/skycaptain/oelint-adv-issue-721. Running oelint-adv either standalone or as pre-commit hook results in the following:

```shell
$ pre-commit run --verbose --all-files oelint-adv
Advanced oelint..........................................................Failed
- hook id: oelint-adv
- duration: 0.45s
- exit code: 1

Loaded rules:
        oelint.append.protvars
        oelint.append.protvars.LICENSE
        oelint.append.protvars.LIC_FILES_CHKSUM
        oelint.append.protvars.PR
        oelint.append.protvars.PV
        oelint.append.protvars.SRCREV
        oelint.bbclass.underscores
        oelint.exportfunction.dash
        oelint.file.inactiveupstreamdetails
        oelint.file.inappropriatemsg
        oelint.file.includenotfound
        oelint.file.includerelpath
        oelint.file.inlinesuppress_na
        oelint.file.nospaces
        oelint.file.patchsignedoff
        oelint.file.requireinclude
        oelint.file.requirenotfound
        oelint.file.underscores
        oelint.file.upstreamstatus
        oelint.func.specific
        oelint.newline.consecutive
        oelint.newline.eof
        oelint.spaces.emptyline
        oelint.spaces.linebeginning
        oelint.spaces.linecont
        oelint.spaces.lineend
        oelint.tabs.notabs
        oelint.task.addnotaskbody
        oelint.task.customorder
        oelint.task.dash
        oelint.task.docstrings
        oelint.task.heredocs
        oelint.task.multifragments
        oelint.task.network
        oelint.task.noanonpython
        oelint.task.nocopy
        oelint.task.nomkdir
        oelint.task.nopythonprefix
        oelint.task.order
        oelint.task.order.do_build
        oelint.task.order.do_compile
        oelint.task.order.do_configure
        oelint.task.order.do_fetch
        oelint.task.order.do_install
        oelint.task.order.do_package
        oelint.task.order.do_patch
        oelint.task.order.do_populate_sysroot
        oelint.task.order.do_unpack
        oelint.task.pythonprefix
        oelint.var.addpylib
        oelint.var.badimagefeature
        oelint.var.badimagefeature.allow-empty-password
        oelint.var.badimagefeature.allow-root-login
        oelint.var.badimagefeature.dbg-pkgs
        oelint.var.badimagefeature.debug-tweaks
        oelint.var.badimagefeature.dev-pkgs
        oelint.var.badimagefeature.eclipse-debug
        oelint.var.badimagefeature.empty-root-password
        oelint.var.badimagefeature.post-install-logging
        oelint.var.badimagefeature.ptest-pkgs
[... 220 lines cut ...]
        oelint.vars.pnusagediscouraged
        oelint.vars.pythonpnusage
        oelint.vars.renamed
        oelint.vars.sectionlowercase
        oelint.vars.spacesassignment
        oelint.vars.specific
        oelint.vars.srcuriappend
        oelint.vars.srcurichecksum
        oelint.vars.srcuridomains
        oelint.vars.srcurifile
        oelint.vars.srcurigittag
        oelint.vars.srcurioptions
        oelint.vars.srcurisrcrevtag
        oelint.vars.summary80chars
        oelint.vars.summarylinebreaks
        oelint.vars.unpackdir
        oelint.vars.valuequoted
        oelint.vars.virtual
Parsing [redacted]/meta-oelint-adv-issue-721/conf/layer.conf
Parsing [redacted]/meta-oelint-adv-issue-721/conf/layer.conf
Parsing [redacted]/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:warning:oelint.vars.bbvars.PROVIDES:Variable 'PROVIDES' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:5:warning:oelint.vars.bbvars.PACKAGES:Variable 'PACKAGES' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:9:warning:oelint.vars.bbvars.RDEPENDS_packagegroup-mygroup:Variable 'RDEPENDS_packagegroup-mygroup' should be set on a disto/layer or local.conf level, not in a recipe [branch:true]
```

priv-kweihmann, 2025-03-03:
Still no luck on my end, as I only get the valid reports of 

```
/oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:error:oelint.var.mandatoryvar.DESCRIPTION:Variable 'DESCRIPTION' should be set [branch:true]
/oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:info:oelint.var.suggestedvar.BUGTRACKER:Variable 'BUGTRACKER' should be set [branch:true]
/oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:info:oelint.var.suggestedvar.SECTION:Variable 'SECTION' should be set [branch:true]
/oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:error:oelint.var.mandatoryvar.SUMMARY:Variable 'SUMMARY' should be set [branch:true]
/oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:9:warning:oelint.vars.mispell:'RDEPENDS_' is unknown, maybe you meant 'RDEPENDS' [branch:true]
```

reproduction:

```
docker run -it python:3.12 /bin/bash
pip3 install oelint-adv
git clone https://github.com/skycaptain/oelint-adv-issue-721
oelint-adv /oelint-adv-issue-721/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb
```

**NOTE**: any other supported python version reports the same (tested 3.9, 3.10. 3.12, 3.13)

BUT if I use python 3.14-rc as a base, I can see your reported issues.
As far as I digged into the issue, this is a python bug (somewhere located in oelint-parser).

Could you please confirm that you dont' see the reported issues on any of the currently supported python versions?

priv-kweihmann, 2025-03-03:
Found the issue - it's a breaking API change in python's multiprocessing code (https://docs.python.org/3.14/library/multiprocessing.html#contexts-and-start-methods) - will provide a fix

priv-kweihmann, 2025-03-03:
and thanks for highlighting that one, even if python 3.14 is not yet officially supported - really started scratching my head about that one for quite a while

skycaptain, 2025-03-03:
Great catch. Sorry for not mentioning the Python version and OS initially. I'm on macOS where I can reproduce the issue with Python 3.11, 3.12, 3.13 and 3.14.0a5 (haven't tested other versions yet), e.g.:

```shell
$ pyenv install 3.12
python-build: use openssl@3 from homebrew
python-build: use readline from homebrew
Downloading Python-3.12.9.tar.xz...
-> https://www.python.org/ftp/python/3.12.9/Python-3.12.9.tar.xz
Installing Python-3.12.9...
python-build: use readline from homebrew
python-build: use ncurses from homebrew
python-build: use zlib from xcode sdk
Installed Python-3.12.9 to [redacted].pyenv/versions/3.12.9
$ pyenv global 3.12
$ pipx install oelint-adv --python=python3.12
  installed package oelint-adv 7.2.1, installed using Python 3.12.9
  These apps are now globally available
    - oelint-adv
done! ✨ 🌟 ✨
$ oelint-adv ...
...
```

According to the Python docs, `spawn` appears to be the default start method here (as it is on Windows):

> Changed in version 3.8: On macOS, the spawn start method is now the default. The fork start method should be considered unsafe as it can lead to crashes of the subprocess as macOS system libraries may start threads. See bpo-33725.

priv-kweihmann, 2025-03-03:
> Great catch. Sorry for not mentioning the Python version and OS initially. I'm on macOS where I can reproduce the issue with Python 3.11, 3.12, 3.13 and 3.14.0a5 (haven't tested other versions yet), e.g.:
> 
> $ pyenv install 3.12
> python-build: use openssl@3 from homebrew
> python-build: use readline from homebrew
> Downloading Python-3.12.9.tar.xz...
> -> https://www.python.org/ftp/python/3.12.9/Python-3.12.9.tar.xz
> Installing Python-3.12.9...
> python-build: use readline from homebrew
> python-build: use ncurses from homebrew
> python-build: use zlib from xcode sdk
> Installed Python-3.12.9 to [redacted].pyenv/versions/3.12.9
> $ pyenv global 3.12
> $ pipx install oelint-adv --python=python3.12
>   installed package oelint-adv 7.2.1, installed using Python 3.12.9
>   These apps are now globally available
>     - oelint-adv
> done! ✨ 🌟 ✨
> $ oelint-adv ...
> ...
> According to the Python docs, `spawn` appears to be the default start method here (as it is on Windows):
> 
> > Changed in version 3.8: On macOS, the spawn start method is now the default. The fork start method should be considered unsafe as it can lead to crashes of the subprocess as macOS system libraries may start threads. See bpo-33725.

Okay that explains it better - let me know if version 7.2.2 works for you now

closed 2025-03-03 by priv-kweihmann

skycaptain, 2025-03-03:
Nope. oelint-adv now even crashes:

```
OOPS - That shouldn't happen: context has already been set - ['meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb']
Traceback (most recent call last):
  File "[redacted]/.local/pipx/venvs/oelint-adv/lib/python3.13/site-packages/oelint_adv/__main__.py", line 119, in main
    issues = run(args)
  File "[redacted]/.local/pipx/venvs/oelint-adv/lib/python3.13/site-packages/oelint_adv/core.py", line 498, in run
    mp.set_start_method('fork')  # pragma: no cover
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.2/Frameworks/Python.framework/Versions/3.13/lib/python3.13/multiprocessing/context.py", line 247, in set_start_method
    raise RuntimeError('context has already been set')
RuntimeError: context has already been set
```

skycaptain, 2025-03-03:
Looks good:

```
$ oelint-adv meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb
Loaded rules:
        oelint.append.protvars
        oelint.append.protvars.LICENSE
        oelint.append.protvars.LIC_FILES_CHKSUM
        oelint.append.protvars.PR
        oelint.append.protvars.PV
        oelint.append.protvars.SRCREV
        oelint.bbclass.underscores
        oelint.exportfunction.dash
        oelint.file.inactiveupstreamdetails
        oelint.file.inappropriatemsg
        oelint.file.includenotfound
        oelint.file.includerelpath
        oelint.file.inlinesuppress_na
        oelint.file.nospaces
        oelint.file.patchsignedoff
        oelint.file.requireinclude
        oelint.file.requirenotfound
        oelint.file.underscores
        oelint.file.upstreamstatus
        oelint.func.specific
        oelint.newline.consecutive
        oelint.newline.eof
        oelint.spaces.emptyline
        oelint.spaces.linebeginning
        oelint.spaces.linecont
        oelint.spaces.lineend
        oelint.tabs.notabs
        oelint.task.addnotaskbody
        oelint.task.customorder
        oelint.task.dash
        oelint.task.docstrings
        oelint.task.heredocs
        oelint.task.multifragments
        oelint.task.network
        oelint.task.noanonpython
        oelint.task.nocopy
        oelint.task.nomkdir
        oelint.task.nopythonprefix
        oelint.task.order
        oelint.task.order.do_build
        oelint.task.order.do_compile
        oelint.task.order.do_configure
        oelint.task.order.do_fetch
        oelint.task.order.do_install
        oelint.task.order.do_package
        oelint.task.order.do_patch
        oelint.task.order.do_populate_sysroot
        oelint.task.order.do_unpack
        oelint.task.pythonprefix
        oelint.var.addpylib
        oelint.var.badimagefeature
        oelint.var.badimagefeature.allow-empty-password
        oelint.var.badimagefeature.allow-root-login
        oelint.var.badimagefeature.dbg-pkgs
        oelint.var.badimagefeature.debug-tweaks
        oelint.var.badimagefeature.dev-pkgs
        oelint.var.badimagefeature.eclipse-debug
        oelint.var.badimagefeature.empty-root-password
        oelint.var.badimagefeature.post-install-logging
        oelint.var.badimagefeature.ptest-pkgs
        oelint.var.badimagefeature.serial-autologin-root
        oelint.var.badimagefeature.staticdev-pkgs
        oelint.var.badimagefeature.tools-debug
        oelint.var.bbclassextend
        oelint.var.filesoverride
[... 213 lines cut ...]
        oelint.vars.pkgspecific.pkg_prerm
        oelint.vars.pnbpnusage
        oelint.vars.pnusagediscouraged
        oelint.vars.pythonpnusage
        oelint.vars.renamed
        oelint.vars.sectionlowercase
        oelint.vars.spacesassignment
        oelint.vars.specific
        oelint.vars.srcuriappend
        oelint.vars.srcurichecksum
        oelint.vars.srcuridomains
        oelint.vars.srcurifile
        oelint.vars.srcurigittag
        oelint.vars.srcurioptions
        oelint.vars.srcurisrcrevtag
        oelint.vars.summary80chars
        oelint.vars.summarylinebreaks
        oelint.vars.unpackdir
        oelint.vars.valuequoted
        oelint.vars.virtual
Parsing [redacted]/meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:error:oelint.var.mandatoryvar.DESCRIPTION:Variable 'DESCRIPTION' should be set [branch:true]
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:1:error:oelint.var.mandatoryvar.SUMMARY:Variable 'SUMMARY' should be set [branch:true]
meta-oelint-adv-issue-721/recipes-core/packagegroups/packagegroup-mygroup.bb:9:warning:oelint.vars.mispell:'RDEPENDS_' is unknown, maybe you meant 'RDEPENDS' [branch:true]
```

priv-kweihmann, 2025-03-03:
Nice!
Btw as you were running only with the standard set of variables and none of the release specific extensions, expect some new warnings in your code base (changes from at least back in time of oelint-adv 5.x were never applied to your setup, because of this bug)

commit 037f3a15 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
commit 26262f88 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
commit 222039f1 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
PR #724 (2025-03-03): core: enforce fork as mp context
as we need to share all resources of the linter's
main process with the child tasks running the checks. Otherwise things like the CONSTANTS interface will be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722